goodtft/LCD-show #235: Screen goes white after updating Raspberry Pi OS

danielktdoranie, 2020-08-20:
Screen goes white after updating Raspberry Pi OS. Doing sudo apt update then sudo apt upgrade, sudo reboot, when the Raspberry Pi reboots the display goes white again.

plawer, 2020-08-23:
Same problem here.

2000jago, 2020-08-25:
Same problem here.   
Is anyone looking into this?

danielktdoranie, 2020-08-25:
> Same problem here.
> Is anyone looking into this?

Doesn’t seem like it, no.

mattglg, 2020-08-25:
A fix from another site is...

1. Executing apt-get upgrade will cause the LCD to fail to work properly. In this case, you need to edit the config.txt file in the SD card and delete this sentence: dtoverlay=ads7846.

https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)

Might be worth a try.

2000jago, 2020-08-26:
> edit the config.txt file in the SD card and delete this sentence: dtoverlay=ads7846.

My config.txt file does not contain this statement before (or after) the apt-get update/upgrade.  The only (uncommented) line that references dtoverlay is `dtoverlay=tft35a:rotate=270` but deleting that made no difference.

danielktdoranie, 2020-08-26:
The support on this product is big poo poo

cschweers, 2020-08-26:
Let me summarize: Everything worked, then you did something stupid and broke it and now the support is bad?

This is an open source project! 
Download the source code, fix the issue and create a pull request. So other people can benefit from your effort! :-)

HunterAhlquist, 2020-08-26:
> Let me summarize: Everything worked, then you did something stupid and broke it and now the support is bad?
> 
> This is an open source project!
> Download the source code, fix the issue and create a pull request. So other people can benefit from your effort! :-)

"did something stupid"
yeah man, updating your operating system is stupid.
"download the source and fix it yourself"
not everyone here is a developer, consumers uses raspberry pi too.

This has been fixed anyway. Follow the same instructions here, but replace this repo with https://github.com/MrYacha/LCD-show.git
At least that works for me.

2000jago, 2020-08-26:
Not sure I would consider doing and apt-get update/upgrade as doing
"something stupid" though.  We didn't break it deliberately.

If we had the skills to fix it ourselves we wouldn't be here looking for
help from someone who clearly does have the skills to fix it.

Also, I don't think asking for some input from the projects creator is
completely unreasonable. After all that is what the "issues" section is
for.  Reporting issues to the dev(s) so that "issues" can be fixed.

We all understand this is an open source project and we are at the mercy of
the creator for support.  Fair enough.
If no support (appears to be) forthcoming, it's reasonable for people who
depend on this project to be disappointed.

Hopefully a fix or a workaround will come soon.

cschweers, 2020-08-26:
> "did something stupid"
> yeah man, updating your operating system is stupid.
if you don't make a backup and don't know how to roll back what you do, than clearly yes ;-)
This is a true Peter Parker moment ... "With great power comes great responsibility"

To be honest, my answer was primarily directed to the OP.  I am of the opinion that dissatisfaction in no case entitles the use of offensive or obscene language.

studiosacchetti, 2020-08-26:
> "did something stupid"
> yeah man, updating your operating system is stupid.
> "download the source and fix it yourself"
> not everyone here is a developer, consumers uses raspberry pi too.
>
> This has been [...]

it work fro me too!!

2000jago, 2020-08-26:
> if you don't make a backup and don't know how to roll back what you do, than clearly yes ;-)

Sounds like back-peddling to me.
There can be many valid reasons for not being able OR willing to roll back a change.  Performing a routine update with the expectation that it will not break things is not unreasonable.  If/when it does and you are not able or willing to roll back is also not considered "doing something stupid".  
It's also not unreasonable to assume that changes to a dependency should follow a kernel update if there are issues, not the other way around.  I'd rather have a secure kernel with no working screen than a vulnerable kernel that keeps my display driver happy, so suggesting one should roll back an important update to solve a display issue that should be attended to by it's dev (or whoever else has the skills) is just pointless.  It's not the core update that needs fixing, it's this repo.

Finally:

> This has been fixed anyway. Follow the same instructions here, but replace this repo with https://github.com/MrYacha/LCD-show.git
> At least that works for me.

Works for me too.  Thank you.

danielktdoranie, 2020-08-26:
> This has been fixed anyway. Follow the same instructions here, but replace this repo with https://github.com/MrYacha/LCD-show.git
> At least that works for me.

Unfortunately, this did not work for me. I have the "MHS35" variant (at least this is what my product manual says, LOL).

Here's the steps I took.

1.) Used Raspberry Pi Imager and chose "Raspberry Pi OS (32-bit)".
2.) Imaged NEW SD card.
3.) Booted Pi 4 Model B 8GB w/ SD Card inserted. Configured. Updated as apart of the install process. 
4.) `git clone https://github.com/MrYacha/LCD-show.git`
5.) `chmod -R 755 LCD-show`
6.) `cd LCD-show/`
7.) `sudo ./MHS35-show`

Raspberry Pi rebooted, the 3.5 touch screen stayed white. I gave it 10 minutes. Disconnected the USB-C power. Plugged in the mini HDMI and now my monitor is displaying super low res (I guess the intended resolution for the 3.5 touch screen?)

So yeah, to warn others, I bought this model:

https://www.amazon.co.uk/gp/product/B07ZQCM57G/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

The supplied manual includes no contact for support (phone or email), just provided this link http://www.lcdwiki.com/MHS-3.5inch_RPi_Display

I am seeing there are multiple github sites for these screens (and of course multiple versions). In short, I wouldn't recommend buying one of these. Luckily I bought with Amazon.co.uk and I have 30 days to return so I will hold out for a little while and see what happens and if it's the same deal a week or 2 from now, I'll just return it.

I like the concept, from my limited use of it it seems like 3.5 inches is just too small for a desktop OS anyway but if it worked with some Android distro, it'd be cool. I was hoping to use mine to output stats like CPU temp, RAM usage, etc. and then use it as a screen to watch films and TV on.

If anyone gets a fix for the MHS35 model post it!

cschweers, 2020-08-27:
> > if you don't make a backup and don't know how to roll back what you do, than clearly yes ;-)
> 
> Sounds like back-peddling to me.

I don't think so! Let's see ....

> There can be many valid reasons for not being able OR willing to roll back a change. Performing a routine update with the expectation that it will not break things is not unreasonable. If/when it does and you are not able or willing to roll back is also not considered "doing something stupid".

This might be a valid assumption for a branded PC with a paid Major OS and a valid service plan. However, even this is not true:
<https://www.forbes.com/sites/gordonkelly/2020/01/27/microsoft-windows-10-warning-security-update-crash-boot-fail-upgrade-windows-10/>

But we're talking about a SBC, build by a non-profit organization. It has an excellent price/performance ratio, is designed for beginners and makers and has mainly an educational purpose. Instead of buying supplies from the raspberry foundation (which surely has been tested) to support the educational purpose, people grep cheapo hardware from the shady corners of the internet. 
In such an case it is negligent to belief that any QA has been taken place! Therefore backups are mandatory before doing an update which might break things. If you don't do that, you act stupid ... qed

> It's also not unreasonable to assume that changes to a dependency should follow a kernel update if there are issues, not the other way around. I'd rather have a secure kernel with no working screen [...]

But it is also true that you need to check _before_ the update to see if the dependencies have been updated to meet the new requirements.

Highpuff, 2020-08-27:
Got the same result with my mhs-35 after being lazy and not checking candidates for update when doing it. Fixed it by doing:

- Download MrYacha's version with 'sudo wget https://github.com/MrYacha/LCD-show/raw/master/usr/waveshare35a.dtbo /boot/overlays/'
- Edit /boot/config.txt and remove old reference of 'dtoverlay=mhs35:rotate=90' and replace it with 'dtoverlay=waveshare35a:rotate=90'.
- Reboot and it worked (or just restart lightdm)

2000jago, 2020-08-27:
> I don't think so! Let's see ....

Nope, it's still back peddling.  Just now poorly obfuscated in a veil of rhetoric.

> This might be a valid assumption for a branded PC with a paid Major OS and a valid service plan. However, even this is not true:
> https://www.forbes.com/sites/gordonkelly/2020/01/27/microsoft-windows-10-warning-security-update-crash-boot-fail-upgrade-windows-10/

Irrelevant to the matter at hand.

> In such an case it is negligent to belief that any QA has been taken place!

You probably mean "believe", not "belief", but I digress.  
It seems unlikely to me that anyone using an open source project does so believing that QA is part of the project or that any assurances or guarantees regarding the intended functionality of the project are being made.  We all know there are risks involved and more often than not, things do not work as expected.  
Once again, there may be valid reasons for not wanting to roll back a kernel upgrade to get a video driver that is lagging behind current kernel versions to work.  Choosing to do an (almost) mandatory kernel update can never be branded as "doing something stupid", it is a completely legitimate and valid action.  
Whether or not the user has preemptively researched future compatibility OR whether they have prepared backups or not does not have anything to do with whether or not choosing to do a kernel upgrade can be referred to as "doing something stupid".

Besides, nobody here has even suggested that they did NOT have backups and did NOT roll back.  That's what I did at first but later decided I'd rather have a white screen and use SSH than a working screen and an insecure kernel.  Your senseless comment is based on pure assumption.

> Let me summarize: Everything worked, then you did something stupid and broke it and now the support is bad?

Seems to me you are now trying to deflect your comment into saying that "not backing up" is the action you were calling "stupid", but your attempt to segue your original statement into another is wasted on me.  That is NOT what you said above.

Perhaps you should word your wise-ass comments better in the future to make it easier for yourself to justify them later.  What you were really trying to say was a rehash of the famous "how dare you complain about something you're getting for free" which is now as old as the internet itself.  

Find another hobby, I'm done with you.

HunterAhlquist, 2020-08-28:
im laughing so hard, this is the last place i expected an internet argument of this kind to break out. god i love the internet

danielktdoranie, 2020-08-28:
> Got the same result with my mhs-35 after being lazy and not checking candidates for update when doing it. Fixed it by doing:
> 
> * Download MrYacha's version with 'sudo wget https://github.com/MrYacha/LCD-show/raw/master/usr/waveshare35a.dtbo /boot/overlays/'
> * Edit /boot/config.txt and remove old reference of 'dtoverlay=mhs35:rotate=90' and replace it with 'dtoverlay=waveshare35a:rotate=90'.
> * Reboot and it worked (or just restart lightdm)
> 
> Of course you have to edit the rotate value to your own need/previous value.

Cheers for your reply, I will try this work around this weekend and report back. Are you running Raspberry Pi OS 32-bit? Are you doing the all the other steps, rebooting and then doing this fix? If so how do you get your monitor to work properly after running the failed scrip? 

God damn this thing is frustrating.

danielktdoranie, 2020-08-28:
> What you were really trying to say was a rehash of the famous "how dare you complain about something you're getting for free" which is now as old as the internet itself.

Yep. I paid for this POS display. Didn’t realise I would have to get a degree in Computer Science and write the driver in order to use it.

On my other Pi 4 I have an Argon One case. It comes with script for fan speed, that keeps working regardless of how many times I apt update && apt upgrade.

tux1c, 2020-08-28:
https://github.com/tux1c/wavesharelcd-64bit-rpi

danielktdoranie, 2020-08-28:
> https://github.com/tux1c/wavesharelcd-64bit-rpi

Thanks but didn't work for me.... however it did change from a white screen... it displayed some boot text and now it's black with white character "_" on screen

ScriptedSolutions, 2020-08-28:
I bought this screen: https://smile.amazon.com/gp/product/B07XQ5B18J/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

Like many of you I had the problem of a white screen being displayed after an update. I wound up having to install the MrYacha branch and use the LCD35-show script to get a response from the screen.

It also worked on the Waveshare 4 inch screen that I hadn't been able to get working. The display comes on, but the touchscreen does not respond. I'll dig into this more and see if I can get it working.

Special thanks to danielktdoranie and Highpuff.

wanthalf, 2020-08-29:
This driver does not work with latest Raspbian, using kernel 5.4. Older version (e.g. 4.19) seems to be supported only.

danielktdoranie, 2020-08-29:
This page here has pre-made images for the Raspberry Pi using the "MHS-3.5inch RPi Display" ... the "Raspbian" image works but when you update it, it stops working. I am going to try the Kali image now. Not a huge fan of prebuilt images for obvious security reasons. 

The value of the Raspbian image is at least you can test the functionality of the hardware. It's been 9 days since I made my original post and we've had nothing, not even an acknowledgement from the developer/manufacturer. This is a warning to anyone thinking of buying one of these: no support and you're on your own.

wanthalf, 2020-08-29:
The above mentioned repository at https://github.com/MrYacha/LCD-show includes a commit called "Edit dtb image to make it work on recent 5.4 kernels". I suppose that is the change missing here.

danielktdoranie, 2020-08-29:
Personally, I'm looking for a 3.5 inch display/case that has better support, the official Raspberry Pi Display is 7 inches.... that is going to have good support I'd think so I may get it and one of those cases.

danielktdoranie, 2020-08-29:
> The above mentioned repository at https://github.com/MrYacha/LCD-show includes a commit called "Edit dtb image to make it work on recent 5.4 kernels". I suppose that is the change missing here.

Good find.

wanthalf, 2020-08-29:
However, it only concerns the driver for 3.5 inch displays. The tft35 driver is beeing replaced with another one from WaveShare. I am going to try with the MHS40 display, but I do not expect success there.

wanthalf, 2020-08-29:
As expected: no change for MHS40 by MrYacha. However, WaveShare has updated drivers for kernel 5.4 for all their displays in their repository at https://github.com/waveshare/LCD-show - running their "./LCD4C-show" makes the "noname" MHS40 display work as well. (Have not tested desktop and touch yet). So, using their overlays is probably currently the only solution.

wanthalf, 2020-08-29:
OK, so the WaveShare repository works for desktop and touch as well. But it probably uses different approach than this repository - I had to start with a clean Raspbian image to make it work, because the Xserver did not want to start with the rests of configurations made before by the scripts from this repository (I had no time to do more research on the details yet).

TheAlienDrew, 2020-08-30:
Getting the same issue here too.

TheAlienDrew, 2020-08-30:
goodtft should just merge the changes MrYacha made

wanthalf, 2020-08-30:
MrYacha just replaced the old driver with a newer one from WaveShare. WaveShare drivers have been updated to work with kernel 5.4. You can use their repository mentioned above.

nathjyo, 2020-09-02:
This was the only method that worked for me

https://github.com/tux1c/wavesharelcd-64bit-rpi

screen = 3.5 inch MPI3501 Driver=ILI9486
latest raspberry pi OS
Pi 4b

closed 2020-09-02 by danielktdoranie