goodtft/LCD-show #236: long touch as right click

rickynet, 2020-08-27:
hi, witch file i must edit for enable this, i have raspbian buster i edited 40-libinput.conf and add edited to be like this
Section "InputClass"
        Identifier "libinput touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "libinput"
        Option "EmulateThirdButton" "1"
        Option "EmulateThirdButtonTimeout" "750"
        Option "EmulateThirdButtonThreshold" "50"
EndSection

but dosnt work, also installed server-xorg-input-evdev and edited their files and nothing

eat-sleep-code, 2020-09-18:
How do I enable right click?
You can emulate a right click with a setting change. Just:

`sudo nano /etc/X11/xorg.conf`
Paste in:

```
Section "InputClass"
   Identifier "calibration"
   Driver "evdev"
   MatchProduct "FT5406 memory based driver"

   Option "EmulateThirdButton" "1"
   Option "EmulateThirdButtonTimeout" "750"
   Option "EmulateThirdButtonMoveThreshold" "30"
EndSection
```
Hit CTRL+X and y to save. Then:

`sudo reboot`
Once enabled, right click works by pressing and holding the touchscreen and will be activated after a short delay.

Source: https://www.raspberrypi.org/documentation/hardware/display/troubleshooting.md

This is tested and working with the latest drivers for MHS35.

rickynet, 2020-09-21:
hi thanks for asnwer i dont have xorg.conf, i hace xorg.conf.d as a directory and files under, 40-libinput.conf (thas is created after install driver), i tried before and didnt work, i try now in a fresh install ans still dosnt work, do you know how to enable two finders scrooll