goodtft/LCD-show #239: Any way to use this display with other distros?

alexanderi96, 2020-09-11:
Hi! Some weeks ago I bought a 3.5 tft display, and I'm glad to see that it is now working with the latest kernel.
I was just wondering if it could work with other distros like Manjaro for example.
I tried to follow the same instructions, but after the reboot i just get the shell, without the login page.

t3kg33k, 2020-11-07:
I, too, would also like to use Ubuntu 20.04 server. I truthfully could care less about touch screen and using a DE. I would love to use the 4.0 display as a standard monitor to boot into multi-user mode to the console (text mode).
I have the XPT2048 4.0 controller. Is there a way to install a driver?

t3kg33k, 2021-03-24:
I've not seen any progress on this. Any update?