goodtft/LCD-show #244: ubuntu mate 20.04 raspberry pi 4 mpi3508 hdmi, touch not work

HyGy, 2020-10-09:
Hello,

I checked this out, and I started the ./MPI3508-show but it writes errors, and touch screen still not work after rebooting.

I'm using ubuntu mate 20.04
Here is the output:

hygy@rpibackup:~/Letöltések/LCD-show$ sudo ./MPI3508-show 
2020
need to update touch configuration
dpkg: error processing archive ./xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb (--install):
 a csomag géptípusa (armhf) eltér a rendszertől (arm64)
Hibák történtek a feldolgozáskor:
 ./xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb
cp: cannot stat '/usr/share/X11/xorg.conf.d/10-evdev.conf': No such file or directory
reboot now
Connection to 192.168.1.149 closed by remote host.

glenndrives, 2020-10-16:
What version of Ubuntu Mate 20.04 are you using, the 32bit or 64bit? 

It looks like the drivers here have not been updated for any 64bit OS.

HyGy, 2020-10-18:
The 64 bit version. Is it working with 32 bit version of mate?

> What version of Ubuntu Mate 20.04 are you using, the 32bit or 64bit?
> 
> It looks like the drivers here have not been updated for any 64bit OS.

glenndrives, 2020-10-18:
This driver does work with the 32 bit version of Ubuntu Mate.

moretocome, 2020-11-15:
@HyGy I have finally gotten the touch screen to work on a Raspberry Pi 4 with Ubuntu Mate 64bit by following [this guide](https://blog.victormendonca.com/2020/10/13/how-to-install-waveshare-3.5inch-lcd-on-ubuntu-20.04/) and adapting it to my device with the specific info for my screen size found in this repository.

goodtft, 2020-12-07:
This system is only for raspbain

HyGy, 2021-07-07:
> This system is only for raspbain

So. I still have problems with this.
- I downloaded the ready made image for latest raspbian with mp3508, but it does not work (MPI3508-3.5inch-600x400-Pi4-2021-03-04-raspios-buster-armhf.7z) its starts booting bot does not start the Xwindows
- is downloaded the Raspberry Pi OS with desktop Release date: May 7th 2021, installed it, the LCD works, but there is no touchscreen, so i checked out the LCD-show.git, then run the mpi3508 script, but it does not reboot after thet

So how can I make this touchscreen work with raspberry pi 4, the latest raspbian, or what you provide to download? The lcd part is working as is couse it is a HDMI display. I just need to make the touchscreen working....

HyGy, 2021-07-07:
> @HyGy I have finally gotten the touch screen to work on a Raspberry Pi 4 with Ubuntu Mate 64bit by following [this [...]

Thanks, I'll try it now. I'm downloading the Ubuntu MATE 20.10

HyGy, 2023-02-15:
Dear Glenn Hickman,

Now it is 2023, did you updated the installer to be able to work with newer
distributions, and with 64bit systems?

regards,
 Gábor

Glenn Hickman ***@***.***> ezt írta (időpont: 2020. okt. 19.,
H, 0:28):