goodtft/LCD-show #249: LCD3.5 and Qt5. How?

mauzil, 2020-11-23:
Hi. There is some way to use LCD3.5 with eglfs or some other drivers for Qt5?

goodtft, 2020-12-07:
I'm sorry. There's no way.

dlmiles, 2021-08-14:
It is not clear if @mauzil is asking if they need EGLFS or they need Qt or they need OpenGL APIs.

As per @goodtft Qt5 with EGLFS backend is not available.
Since OpenGL often has specialist silicon to assist graphics acceleration which would not be present via a TFT driven via GPIO pins.  The purpose of the EGLFS backend is to leverage that silicon.

Using Qt with LinuxFB backend driver might work ?  https://doc.qt.io/qt-5/embedded-linux.html

If you are after OpenGL APIs, it maybe possible to use OpenGL via a 100% software solution for the API like Mesa3D.

mauzil, 2021-08-14:
Hi @dlmiles. I would like to use LCD3.5 directly from Qt5. 
When I posted the question I was working with elgfs.
Perhaps the solution would be to produce a framebuffer driver for direct control of the display (and touchscreen) and then use mesa as a 3d "acceleration".
Maybe in the future I'll try

dlmiles, 2021-08-14:
Do you have /dev/fb0 or /dev/fb* already ?  I think you may, as I think that is how graphics with the TFT works for you right now.

So what you need is Qt5 build with the LinuxFB backend already in it.
Research the Qt5 build you have installed, you want the build config make to create Qt5 from Rasppbian (or you distribution you are using), there is probably a way to get the build options out of it somehow, you are looking for "-linuxfb".

Take a look at: https://wiki.qt.io/Native_Build_of_Qt5_on_a_Raspberry_Pi

This pages appears to clues on how to build yourself a custom copy of Qt5 and the page clearly indicates LinuxFB is part of that build.