goodtft/LCD-show #255: LCD-Show gave error com Raspberry OS Lite with no desktop environment

moisesdfelix, 2020-12-08:
Good Nigth, I installed the raspberry os Lite with in the desktop environment and presented the following errors. After the restart a crash occurs 

`p: cannot stat '/usr/share/X11/xorg.conf.d/99-fbturbo.conf': No such file or directory
cp: cannot create regular file '/usr/share/X11/xorg.conf.d/99-fbturbo.conf': No such file or directory
cp: cannot create regular file '/usr/share/X11/xorg.conf.d/99-fbturbo.conf': No such file or directory
cp: cannot create regular file '/usr/share/X11/xorg.conf.d/99-fbturbo.conf': No such file or directory`

```
need to update touch configuration
Selecting previously unselected package xserver-xorg-input-evdev.
(Reading database ... 45315 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-input-evdev_2.10.5-1_armhf.deb ...
Unpacking xserver-xorg-input-evdev (1:2.10.5-1) ...
Processing triggers for man-db (2.8.5-2) ...
dpkg: dependency problems prevent configuration of xserver-xorg-input-evdev:
 xserver-xorg-input-evdev depends on libevdev2 (>= 0.9.1); however:
  Package libevdev2 is not installed.
 xserver-xorg-input-evdev depends on libmtdev1 (>= 1.1.0); however:
  Package libmtdev1 is not installed.
 xserver-xorg-input-evdev depends on xorg-input-abi-24; however:
  Package xorg-input-abi-24 is not installed.
 xserver-xorg-input-evdev depends on xserver-xorg-core (>= 2:1.18.99.901); however:
  Package xserver-xorg-core is not installed.

dpkg: error processing package xserver-xorg-input-evdev (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 xserver-xorg-input-evdev
reboot now

```

Jemsbsg, 2020-12-15:
Hi !
Have you solved your problem ?

I have exactly the same problem with octoprint

each time i tri to activate the screen, the rapsberry reboot and is unusable, i have to format it.

Thx in advance.

moisesdfelix, 2020-12-16:
I managed to solve by installing the KILI distro module at the link below: 
http://www.lcdwiki.com/res/MHS3528/kali-linux/LCD-show.tar.gz

and then, I ran an apt-get update, and followed the steps:

`sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./MHS35-show`

moisesdfelix, 2020-12-16:
It worked in raspbian with Gui, in the terminal I haven't tested it yet.