goodtft/LCD-show #258: Pi4 does not boot after LCD35-show under ARM64 Raspbian.

red-scorp, 2020-12-27:
P4 does not boot after LCD35-show under ARM64 Raspbian (2020-08-20-raspios-buster-arm64.img).
Literally Pi4 can't boot afterwards. It seams tft35a-overlay.dtb is incorrect.
Has anybody a workaround? How can I enable the 3.5ich LCD?
Thanks in advance!

spuzzelsnest, 2021-01-04:
Hi, 
As the /boot/config.txt is rewritten during the install process, you need to check for these lines in config.txt

`[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2
# 64bit kernel for Raspberry Pi 4 is called kernel8l (armv8a)
kernel=kernel8l-alt.img
[all]
# Tell firmware to go 64bit mode.
arm_64bit=1
`
After installing this package, a few other things need to be chekced.
I found these to work in my ARM64 - 3.5 LCD Touch

`
hdmi_force_hotplug=1
display_rotate=2
dtparam=i2c_arm=on
dtparam=spi=on
enable_uart=1
dtoverlay=mhs35:rotate=270
hdmi_group=1
hdmi_mode=4
hdmi_mode=4
hdmi_cvt 480 320 60 6 0 0 0
hdmi_drive=2
`
This is not the only part of the journey of getting it installed on ARM64, I got it working but it cost me a couple of hours.

Cheers

anthony76-newbi, 2021-01-14:
hi there I'm using a rpi4 8gb I've persisted for a week trying so much on a 35gb sd card it never worked today i tried with a 256gb sd and  finally the lcd is working but I don't know if this helps, I'm still trying to figure out how to get touch screen working now!

Daedalus706, 2021-01-21:
> Hi,
> As the /boot/config.txt is rewritten during the install process, you need to check for these lines in config.txt
> 
> [pi4]
> 
> * 64bit kernel for Raspberry Pi 4 is called kernel8l (armv8a)
>   kernel=kernel8l-alt.img
> * Tell firmware to go 64bit mode.
>   arm_64bit=1
> 
> After installing this package, a few other things need to be chekced.
> I found these to work in my ARM64 - 3.5 LCD Touch
> 
> hdmi_force_hotplug=1
> dtparam=i2c_arm=on
> dtparam=spi=on
> enable_uart=1
> dtoverlay=mhs35:rotate=270
> hdmi_group=1
> hdmi_mode=4
> hdmi_mode=4
> hdmi_cvt 480 320 60 6 0 0 0
> hdmi_drive=2
> 
> This is not the only part of the journey of getting it installed on ARM64, I got it working but it cost me a couple of hours.
> 
> Cheers

I have the same problem and I tried this but it didn't work. The display, in fact, worked but the pi stopped responding before the  boot process was complete. I'm new to Linux and raspberry pis so I have absolutely no idea what else I could do to make it work.

ChavasRegal, 2021-02-07:
I have the same problem, once installed the the pi reboots but the green LED flashes 7 times indicating that the boot kernal cannot be found.

This issue has killed 2 installs.

alexstacey, 2021-05-19:
I got this working on ARM64 with a MHS35. Screen and touch both work. See here for a script https://github.com/alexstacey/MHS35-lcd-64bit-rpi

red-scorp, 2021-05-29:
Thanks all of you. This answers my question very well.

closed 2021-05-29 by red-scorp