goodtft/LCD-show #261: Touchscreen not working

isnotinvain, 2021-01-08:
Display works fine, but touch screen not working. Followed the provided instructions. Using a raspberry pi 3, 5 inch screen, plugged in via usb not via GPIO.

Any ideas for troubleshooting?

Thanks

levelKro, 2021-01-21:
In ma case, I use the MHS35-show to install my 3.5inch with Touchscreen DPI and display work, but not the Touchscreen. It's look like a problem with drivers and latest Raspbian update. You can try an older version (begin of 2020 of before), it's my solution for running display with Retropie.

isnotinvain, 2021-01-27:
@goodtft will you be releasing an update that supports newer versions of Raspbian?

Electronlibre2012, 2021-03-28:
hello all,

so bad @goodtft doesnt give anymore response...

same boat, screen works but not the touch screen...

not good

sngrl, 2022-01-10:
The same problem with Raspberry Pi 3B (latest clean release of the Raspberry Pi OS) + MHS-3.5: screen is working but touchscreen is not

frozenfoxx, 2022-08-03:
Same problem here with TFT35 and GPIO.

luisdavid0202, 2024-03-16:
Same problem here! Any workaround? LCD display image but the touch screen doesn't work.
MHS-3.5” RPi Display-B(MHS35XX)
RPi 3b+ bullseye

kaosbeat, 2024-09-18:
same here on RPi5, boots fine, touch works for 5 seconds and then stops...