goodtft/LCD-show #266: White screen, even after installing the drivers

Sourison, 2021-02-02:
Hi all, 

I've bought this lcd screen. 
https://www.amazon.nl/owootecc-Raspberry-touchscreen-ventilator-koellichaam/dp/B07YV5WYM3/ref=sr_1_1?__mk_nl_NL=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=raspberry+pi+4+touchscreen+3%2C5&qid=1612272149&sr=8-1

After installing the new raspbian os, i've installed the drivers as written below

`sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/`

However, the screen stays white and the hdmi screen keeps freezing on the rainbow screen. 
Did i miss something?

ElKoziRRo, 2021-02-03:
i got the same problemat rpi4 with waveshare 3.5 A  is blank and raspi not boot after install. i guess it`s becouse new boot type and boot config not work.

Denis4l, 2021-05-08:
If you have only done these commands, you did not install the driver. After changing directory ( cd LCD-show ), the installation is performed by launching the program LCDxx-show where xx is the number related to your screen.