goodtft/LCD-show #267: Screen stays white after installing driver

spock34, 2021-02-03:
I have an XPT2046 3.5 inch screen which works, but has always stayed white. I have literally tried everything all the drivers on raspbian and it still stays white. But, when I launch the driver (MHS-35) it changes the resolution of my monitor but doesn't move to the screen. Please fix Ur driver or has anyone got any suggestions? Would help a lot.

stefanfoulis, 2021-02-19:
I have the same problem with the 3.5 inch screen.

stsanford, 2021-02-22:
Same here, I have have tried every which way I can think, but when I reboot, I don't get anything other than a white screen. I have an identical Pi4 that I can swap with, and the screen works, so something borked with the installer script. I have been troubleshooting, but cannot seem to get this resolved. Any guidance would be most appreciated.

spock34, 2021-02-23:
I sent my screen back after thinking it was a fault in the hardware and bought a new one. installed the driver on raspbian, and works perfectly. Got It working on Kali too.

HoldOffHunger, 2022-01-20:
It's 2022, and it's still broken without any reply from the developers.

awnyel, 2022-03-10:
> I sent my screen back after thinking it was a fault in the hardware and bought a new one. installed the driver on raspbian, and works perfectly. Got It working on Kali too.

How you managed to start it on kali I have problems with it

HoldOffHunger, 2022-03-10:
Note: I fixed this by just reflashing my SD card with the newest (bullseye) release from https://www.raspberrypi.com/software/operating-systems/

THEN, proceed with instructions by developer.