goodtft/LCD-show #268: LibreELEC - no command 'git'

GarageMan, 2021-02-06:
Hello, dear developers,
I recently bought the 5inch TFT touch for my Raspberry Pi 3B+ running LibreELEC (latest version)
Unfortunately, I cannot install the drivers from github using the command 'git' (due to not available on LE)

I am lost about how to install the driver to activate the touch-functionality of the display.

Can you please be so kind as to support me on this?

Thanks a lot.

.:. lutz

pheiduck, 2021-02-10:
login into shell via ssh or via hdmi > alt+F3 > login on Shell Prompt and try to install git
sudo apt install git -y

GarageMan, 2021-02-11:
Thanks, pheiduck. I will try that later today and will note my experiences.

GarageMan, 2021-02-13:
Unfortunately, that does not work. it is not possible to install on LibreElec - except using a Docker container.

closed 2021-03-06 by GarageMan

MechWarriorZero, 2023-01-09:
I too having the same issue, I need the driver installed in LibreELEC