goodtft/LCD-show #274: Touch not working

GarageMan, 2021-03-06:
Config:
Raspberry Pi 3B
HifiBerry DAC+ pro
GoodTFT 5 '' connected via GPIO and HDMI
OS: I tested different distributions: volumio, Raspberry OS, Hifiberry OS - all latest versions

The display is working well with the full resolution. Very nice!
(I followed your instructions installing LCD5-show as described on your PDF manual).

Unfortunately, I do not get the touch function to work.
(Mouse and KBD are working well when connected to USB)

What am I doing wrong?

Cheers.

.:. lutz

dlmiles, 2021-08-14:
Same issue here:
Raspberry Pi 4
CUQI MPI7002 (7 inch 1024x600 capacitive touchscreen)

I notice when plugged in I see:
$ lsusb
Bus 001 Device 006: ID 4348:55e0 WinChipHead

Research into this device and it appears to be a general purpose USB capable microcontroller (MCU) see https://github.com/rgwan/librech551
I assume it is in some kind of boot up mode and is awaiting code/blob from host (raspbian Linux) which is code run on it to interface touchscreen with USB and/or provide serial pointer data stream like a mouse.
What is the process/project/mechanism via which the MCU is programmed by the host ?

I also notice this project "LCD-show" has the following:

/etc/modules** includes (are these relevant to my LCD model ?):
```
i2c-dev   # What is this use for (and enablement in boot/config.txt:dtparam=i2c_arm=on)
spi-bcm2835  # What is this used for (and enablement in boot/config.txt:dtparam=spi=on)
flexfb       # This is no longer supported since Kernel 5.4 so it not available by default https://www.raspberrypi.org/forums/viewtopic.php?t=213794
fbtft_device  # This is no longer supported since Kernel 5.4 so it not available by default https://www.raspberrypi.org/forums/viewtopic.php?t=280676
```
Are these required for panel with direct HDMI support ?  A quick look at the info make it look like it is for TFTs that are driven directly from the GPIO of the RaspberryPI.

Re the I2C and SPI does my model need these either ?

This project "LCD-show" should not really enable them (in config.txt) if the specific model being installed does not require them, because it has direct HDMI (for video support) and direct USB (for touchscreen support) and no need to take ownership of GPIOs that maybe being used for some other purpose on the same device.

Showing X.11 not able to find any available "libinput touchscreen catchall" device to make available to use:
```
$ grep Class /var/log/Xorg.0.log
[    11.607] (**) Logitech Wireless Mouse: Applying InputClass "evdev pointer catchall"
[    11.607] (**) Logitech Wireless Mouse: Applying InputClass "libinput pointer catchall"
[    11.607] (**) Logitech Wireless Mouse: Applying InputClass "libinput pointer catchall"
[    11.716] (**) Logitech K540/K545: Applying InputClass "evdev keyboard catchall"
[    11.716] (**) Logitech K540/K545: Applying InputClass "libinput keyboard catchall"
[    11.716] (**) Logitech K540/K545: Applying InputClass "libinput keyboard catchall"
[    11.790] (**) Logitech K540/K545: Applying InputClass "evdev keyboard catchall"
[    11.790] (**) Logitech K540/K545: Applying InputClass "libinput keyboard catchall"
[    11.790] (**) Logitech K540/K545: Applying InputClass "libinput keyboard catchall"
```

salk31, 2021-09-11:
I think I might be in the same boat just getting the usb id for the ch554 microcontroller (also the label on the chip that the "touch" usb socket is connected to).
I had hope when I saw this issue https://github.com/goodtft/LCD-show/issues/10 but no luck for me fiddling around with usb connections. 
I've tried a Pi 3 and a Pi 4 without luck.

vshuraeff, 2022-01-06:
with no connected HIDs like keyboard and mices `lsusb` looks as 
```bash
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

and thats all. no clicks, touch does not works. 
I use `LCD35`

sngrl, 2022-01-10:
The same problem with Raspberry Pi 3B + MHS-3.5: screen is working but touchscreen is not

MsternSC, 2022-07-12:
Raspberry Pi 4 did not work for me, but Raspberry Pi 3B+ does work.