goodtft/LCD-show #278: Raspberrypi - Bluetooth stop working after latest Buster update

pesa1234, 2021-03-14:
Please do not change on cmdline.txt `console=serial0,115200 console=tty1`

In your script this `console=tty1 console=ttyAMA0,115200` cause problem on bluetooth after latest update on Buster.

(Missing bluetooth icon)

elmoi, 2021-03-14:
Same issue here, my rbpi3b+ is working perfect from a new instalation, but as soon I install the LCD5-show script, on the new reboot, raspberry is not able to detect the bluetooth device.

Missing bluetooth icon, and services related to bluetooth fail to start, so other scripts are not working as bluetoothctl, the system doesn't recognize the bluetooth device at all.

pesa1234, 2021-03-14:
Change the line as indicated on my post to have Bluetooth again

elmoi, 2021-03-14:
I have test it right now and works perfectly.

OLD cmdline before screen install: `console=serial0,115200 console=tty1 root=PARTUUID=5a410835-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles`

NEW FAILING cmdline.txt: `dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo`

As @pesa1234 said, in the new cmdline.txt this part `console=ttyAMA0,115200` should be changed by `console=serial0,115200`

This file is located on the partition boot (FAT32), so you have two alternatives:

1. Mount the partition on Linux and modify it: `$ sudo mount /dev/mmcblk0p1 /media` and modify it on dir /media 
2. Or extract the SD and do it from a windows OS. 

After a reboot, the system recovers the bluetooth device. 

Thanks @pesa1234 !!!

Julian-MJK, 2021-03-16:
I'm consistently having this issue when swapping between using HDMI and the touchscreen, with LCD-hdmi and LCD35-show, so it's a consistent issue, and your fix works like a charm every time, thanks!