goodtft/LCD-show #282: Brightness control by software ( MPI7002 )

lukecanny, 2021-04-01:
Is it possible to change brightness by software on the  MPI7002 7" Touchscreen? I read on this page that there is no hardware to realise the software control but wondering if that is the same for this model? Thank you.

almirus, 2021-11-10:
do you found any solution?

DJTommek, 2021-11-10:
Maybe it is offtopic, but for some it might be useful.

My usage of display is as HTML page so I came up with this easy workaround using CSS and Javascript by creating black overlay over page with initial opacity 0 (which is invisible). Then you are able to control brightness via changing opacity from 0 (100% brightness) to 0.8 (20% brightness), via button, websocket, AJAX, etc.

I've created simple demo on Fiddle, https://jsfiddle.net/DJTommek/ovugep7r/ including persisting brightness after page refresh.

lukecanny, 2021-12-23:
> do you found any solution?

No, it is not possible it seems. @almirus 

Maybe @DJTommek 's response is applicable. I have not looked into it.

closed 2023-10-14 by lukecanny