goodtft/LCD-show #290: Raspberry stuck after reboot LCD35-show

kura6789, 2021-06-29:
Hi, my raspberry stuck on line "Started Load/Save RF Kill Switch Status." - it appears in HDMI monitor, 3,5 inch display is still blank, ssh is working. I only changed /dev/sda2 in cmdline.txt. Can u hel me pls? 
[image]

carlni2008, 2021-09-12:
Hi, have you experienced this before you installed your screen. If no, lease try changing your power adapter or use this https://github.com/MrYacha/LCD-show instead. This is a better version of lcd-show

ahnyerkeester, 2023-02-15:
> Hi, have you experienced this before you installed your screen. If no, lease try changing your power adapter or use this https://github.com/MrYacha/LCD-show instead. This is a better version of lcd-show

Though
`git clone https://github.com/goodtft/LCD-show.git`
had worked for me previously, it recently left the screen blank on clean installs so I switched to 
`git clone https://github.com/MrYacha/LCD-show.git`
and that took care of it!