goodtft/LCD-show #296: Octoprint Vs. GoodTFT35

MJBridge, 2021-07-21:
I'm very green when it comes to Raspberry Pi based CLi work.
I am trying to build an Octoprint/Ocotodash unit for my 3D printer.
I have ran the update script to update my build to latest version.

I am running a Raspberry Pi 3B V2.1 with Octoprint 1.6.1 -no Desktop

I have ran the script below using Putty/SSH:

_- sudo rm -rf LCD-show
- git clone https://github.com/goodtft/LCD-show.git
- chmod -R 755 LCD-show
- cd LCD-show/
- sudo ./LCD35-show_

I reboot and the screen goes from white to black and that's all.
Here is a transcript of the error log I get, but it doesn't mean a lot to me at this point other than it has failed on a few items?

Can anyone help please?

~~~~~~~~~~~~~~~~
_(Reading database ... 47027 files and directories currently installed.)
Removing xserver-xorg-input-evdev (1:2.10.6-1+b1) ...
dpkg: warning: while removing xserver-xorg-input-evdev, directory '/usr/share/X1                                                                                                             1/xorg.conf.d' not empty so not removed
Processing triggers for man-db (2.8.5-2) ...
2020
need to update touch configuration
Selecting previously unselected package xserver-xorg-input-evdev.
(Reading database ... 47013 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb ...
Unpacking xserver-xorg-input-evdev (1:2.10.6-1+b1) ...
Processing triggers for man-db (2.8.5-2) ...
dpkg: dependency problems prevent configuration of xserver-xorg-input-evdev:
 xserver-xorg-input-evdev depends on libevdev2 (>= 1.2.2+dfsg-1~); however:
  Package libevdev2 is not installed.
 xserver-xorg-input-evdev depends on libmtdev1 (>= 1.1.0); however:
  Package libmtdev1 is not installed.
 xserver-xorg-input-evdev depends on xorg-input-abi-24; however:
  Package xorg-input-abi-24 is not installed.
 xserver-xorg-input-evdev depends on xserver-xorg-core (>= 2:1.18.99.901); howev                                                                                                             er:
  Package xserver-xorg-core is not installed.

dpkg: error processing package xserver-xorg-input-evdev (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 xserver-xorg-input-evdev
reboot now_

BullzLabz, 2021-07-24:
Do you've tried this :

http://www.fintechcommunications.com/blog/?p=375

galambo, 2021-07-27:
having the exact same issue

galambo, 2021-07-27:
the doc shown here is out of date. there is no SPI option within the advance menu choice. this was working before (had to regen the OS) and then this issue came up. (the original was before the latest update for the driver)