goodtft/LCD-show #298: raspberry pi 4 3.5inch

taraxidas, 2021-09-08:
Hello does the lcd35show work with raspberry pi 4b or will i get a whitescreen?

mkikets99, 2021-10-31:
I think it will. 
I have RPi 3b+ and everything works great. as long as pinout on RPi 4b is not changed this must work

Spartacus1066, 2021-11-14:
Yes, but the start button and right-click produce nothing on it for some reason.

mrfoxie, 2022-01-12:
@taraxidas it will work with raspberry pi 4b but for some reason, start button is not working

pheiduck, 2022-01-12:
I always install the driver like this, otherwise it will not work in my case (RPI3b+)
```
sudo apt install xserver-xorg-input-evdev -y
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
```

mrfoxie, 2022-01-12:
its work fine on rpi4b+ but i have to make some tweaks in config.txt and invert the axis

mrfoxie, 2022-01-13:
> Hello does the lcd35show work with raspberry pi 4b or will i get a whitescreen?

just clone the code and install `LCD35-show` and after that edit 
```
sudo nano /boot/config.txt
```
after entering in the file find for this line at the bottom of the file `dtoverlay=tft35a:rotate=90` comment this line and bellow this line add new dtoverlay
```
dtoverlay=piscreen,speed=16000000,rotate=90
```
save the file and then reboot your pi and it's done ✅