goodtft/LCD-show #301: Cannot find source code for tft35a-overlay.dtb

hsnprojects, 2021-10-09:
I was looking for source code for 3.5" LCD Driver, which seems to be tft35a-overlay.dtb, but I could not locate the source code.
Is this fully open source project?

ChazTuna, 2022-03-20:
Me too - what's with the hidden source code?

bkw777, 2022-03-22:
A .dtb file is equivalent to it's source.
https://community.arm.com/oss-platforms/w/docs/525/device-tree
Generate any .dts you want from the .dtb

Though I agree that the repo should have .dts files, and .dtb files should be in releases, just so that the repo managed files are human readable and sensibly diffable.

paulwratt, 2022-03-22:
at least there is info on this now

I agree with `.dts` but then would be a need to have a convertion tool too. Maybe just some more docs `README-dtb-dts.md` ..

bkw777, 2022-03-22:
I linked to the docs about dtc's . There is no need to guess. Yess the tool is preinstalled and yes it decompiles dtb to dts and back to dts without loss. It's almost the first thing at the top of the page.