goodtft/LCD-show #305: Option for non-xwindows install or simply move terminal

TedMN, 2021-12-02:
Thanks for the utility for config and driver install.  It did not fit my use case but got my 95% there.  I used the minimal raspberry pi install that does not have x11 installed or configured.  So while the driver installed, the dev/fbX was created and /boot/config.txt settings configured etc one thing that was not was /boot/cmdline.txt.

https://ozzmaker.com/enable-console-on-piscreen/

https://www.kernel.org/doc/Documentation/fb/fbcon.txt

In my case I used fbcon=map:10 after the rootwait at the end.  Worked like a charm.

I am happy to contribute a feature patch or it is fairly simple to implement.  Even just an install readme note, I can write that up.  

Best regards.

gustavojm27, 2021-12-14:
Great job!!! Thanks for sharing! Worked like a charm!

skylarweaver, 2021-12-24:
@TedMN can you share what OS you had installed? This fix allowed me to boot to terminal on Ubuntu MATE Desktop 20.10, but I'm trying to use this screen on ubuntu-server 20.4 without a GUI and only ever get a black screen, even with this fix.

Any more insight into your OS and suggestions would be helpful. Thanks

paulwratt, 2021-12-25:
@skylarweaver 

> I used the minimal raspberry pi install that does not have x11 installed or configured. 

thats the "server" edition of RPiOS (ie console only SD-card image)

is-paranoia, 2023-06-26:
@skylarweaver hi! Did you solve the problem with ubuntu server 20.4 (cli only)? Can't find a solution for a long time(

OmerSchechter, 2025-12-24:
> [@skylarweaver](https://github.com/skylarweaver) hi! Did you solve the problem with ubuntu server 22.4 (cli only)? Can't find a solution for a long time(

What @TedMN said is to add `fbcon=map:10` in the /boot/firmware/cmdline.txt file after `rootwait`. It worked for me. Shows the CLI perfectly on the small display. Hope that helps.