goodtft/LCD-show #311: lcd not working

githubsharezone, 2022-01-28:
hi i have 

root@raspberrypi:~# uname -a
Linux raspberrypi 5.10.92-v8+ #1514 SMP PREEMPT Mon Jan 17 17:39:38 GMT 2022 aarch64 GNU/Linux

with 8gb ram and 128GB ssd 

the os is updated but when i install the driver the raspberry not boot , i need remove setting on file confix.txt for entry on device.

can you hel me ?

DRK-512, 2022-02-03:
Hey, try reflashing the device with a new version of raspian. 
After doing that, go into terminal, git clone the repo, chmod755 the directory. 
One important thing to note that guides don't have is you also have to type 
sudo su
before you run the LCD script. The reason for this is because since you are editing the config file, you need more than sudo permission on a raspberry pi for some odd reason. Once you type sudo su then type ./LCDxx-show then it should run the script, reset your device, and the screen should work

lacymorrow, 2022-03-02:
I had to use `sudo su` to run the install, `sudo` by itself didn't work