goodtft/LCD-show #313: LCD35 conflict issue with libcamera

bb554C, 2022-03-01:
Using the 3.5 LCD driver provided as LCD35 causes an issue wherein the camera module stops working. I've read possible issues are that the LCD is interfering with the signal from the camera moduie

Abdhul22, 2022-05-04:
@bb554C @goodtft Facing the same issue.
Has someone figured out how to fix this?

Abdhul22, 2022-05-04:
Camera works before installing the LCD35 Drivers. After installing camera does not work and it's red light in on continuously

bb554C, 2022-05-05:
I reverted back to Legacy OS version. Camera doesn't work due to conflict with the signals with libcamera.

WayneKim92, 2023-09-23:
I got the same issue.

WayneKim92, 2023-09-23:
```
libcamera-hello --qt-preview
```

smathles, 2024-03-14:
Experiencing the same thing with a R/Pi camera module 3 and this driver. It looks like the driver enables the legacy camera option in `sudo raspi-config`, which makes the camera fail but the screen work. Was never able to reconcile both unfortunately.

asherevan, 2024-10-08:
I have the same problem. Is there a way to fix it?

kwatkins, 2024-10-17:
Not only this, it seems to kill the camera on raspberry pi 5. What a pain.

asherevan, 2025-07-06:
I seem to have somehow gotten it to work by disabling the dtoverlay=vc4-kms-v3d line.