goodtft/LCD-show #315: Display not working on the new  Pi Zero 2

brentfraser, 2022-03-06:
I've got two 3.5 inch displays.  They work fine on a Pi Zero 1 and on a Pi 3B.  But not on a Pi 2 W (the new quad core).

The install procedure works and doesn't report any errors, but on reboot the display just shows the back light.
uname -a:
`Linux raspberrypi 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux`

config.txt:
```
# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
hdmi_force_hotplug=1
dtparam=i2c_arm=on
dtparam=spi=on
enable_uart=1
dtoverlay=tft35a:rotate=90
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt 480 320 60 6 0 0 0
hdmi_drive=2
```
cmdline.txt:
`console=serial0,115200 console=tty1 root=PARTUUID=eec0283e-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
`

brentfraser, 2022-03-07:
Never mind.  Looks like it was a bad solder joint on the Pi header.  It works fine on my other Pi Zero 2.

closed 2022-03-07 by brentfraser

Victor-Otsuga, 2024-05-14:
I'm having the same problem but I haven't soldered the pin bar to the Rasp yet, anyway I have an HDMI cable connected to a monitor and after installing the drivers the system doesn't boot, it's possible that it doesn't boot due to the screen not soldered?

cschweers, 2024-05-15:
What does the serial console log say? see https://www.jeffgeerling.com/blog/2021/attaching-raspberry-pis-serial-console-uart-debugging

Victor-Otsuga, 2024-05-15:
> What does the serial console log say? see https://www.jeffgeerling.com/blog/2021/attaching-raspberry-pis-serial-console-uart-debugging

i dont have a serial adapter and time to wait the delivery time, iam trying some type of lucky guess to solve, if you have any hint of what can be based on the info please tell me.

- Raspberry Pi Zero 2 W
- RPI display 3,5inch
- 16gb SD card 
- a lot of OS tested

i think there is a error on lcd drivers instalation ou something like this because in the lcd store with the seller SD the screen worked fine.

brentfraser, 2024-05-15:
It could be just something in your pi's boot/config.txt file related to the various hdmi_* (especially hdmi_force_hotplug and hdmi_drive ?) settings.