goodtft/LCD-show #320: 4 inch display. How to control / slow down extremely annoying fan?

mdrmdrmdr, 2022-05-01:
Just installed the 4 inch display on my Pi 4 for just console mode. Works very nice!

But, the small fan installed on the back side of the display is extremely annoying with a high frequency.

- Is it possible to control the rpm of the fan with software?
- Or is it possible to connect the fan to GPIO's and control it then with software?
- Might it be possible to add a resistor to the 5V cable to slow it down?

Thanks for any tips...

mdrmdrmdr, 2022-05-05:
Got the information from the seller on Amazon, that to slow down the fan, the 0 (zero) Ohm resistor just above the fan connector on the board needs to be replaced with a 20 Ohm resistor. I'll report here if it works.

mowtschan, 2024-08-15:
@mdrmdrmdr could you share some picture of the screen please?

mdrmdrmdr, 2024-08-15:
Currently I do not use the display. So can't show it running. But in the picture you see the soldered header and the 33 Ohm resistor. Hope this helps. 

[image]