goodtft/LCD-show #324: Cannot identify correct 3.5" LCD

ben-barnard, 2022-07-12:
https://www.aliexpress.com/item/2251832419095697.html?spm=a2g0o.order_list.0.0.61c21802OHgSBi&gatewayAdapt=4itemAdapt

I have tried multiple pis (zero, zero w, zero 2 w, 3B+) and all of the different 3.5" drivers here, but none have been successful. I have tried regular Raspbian, Raspbian Lite, I have tried DietPi. Can anybody identify anything unusual about this panel that would make it not work? I'm stumped - the commenters seem to have had no problem.

WermRolent, 2022-12-06:
maybe you install 32 bit system?

theinfinitegirl, 2024-01-18:
Apologies if you've tried this, but have you tried the plain 'LCD35-show' driver? I have the same screen, and that worked perfectly well (none of the 'M' drivers did).

ben-barnard, 2024-01-18:
> Apologies if you've tried this, but have you tried the plain 'LCD35-show' driver? I have the same screen, and that worked perfectly well (none of the 'M' drivers did).

I actually did get it to work with an older version of Raspbian - did you get it to work with a more recent version? What was your performance like?