goodtft/LCD-show #326: kmscube not working after drivers installed

AshleighAdams, 2022-09-25:
I'm trying to get `kmscube` working from Raspian Lite, but when trying, I get the following:

```
drmGetDevices2 failed: No such file or directory
could not open drm device
failed to initialize legacy DRM
```

AshleighAdams, 2022-09-26:
I found the following to enable the HDMI port's (line was commented in my config.txt):

```sh
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
```

Is there no way to get this to work with the TFT?

Simzie, 2023-01-26:
I'm also find this an issue, preventing me from really using the screen on my Zero 2 W. I want to use the preview option with Picamera2 on Bullseye, which requires full KMS, but only get a blank screen. FKMS will load but is not supported by Picamera2, so no preview shows. Does anyone know what might be required to make KMS function?

kristian, 2023-08-16:
Any luck yet by anyone? Also the Midori browsers requires full KMS, but I could not get the TFT working yet.