goodtft/LCD-show #329: LCD7C not working in RPi4 bullseye 64bit

UgaitzEtxebarria, 2022-11-20:
I am trying to upgrade mi old RaspberryPi3b with Buster to a RaspberryPi4b where i have installed new raspbian version (Bullseye). I tried to install the script of this repository but doesnt work. Later, i tried to modify the config.txt file but none of changes I made fix the display.

Does the script work for this hardware and software?

Simzie, 2023-01-26:
I can't speak to the hardware as I've only used it with a 3B+ and Zero 2 W, however in both instances it fails when the DRM driver is used. Have you checked that "overlay=vc4-kms-v3d" is not included in your config.txt? Changing it to "overlay=vc4-fkms-v3d" works for me (or disable completely for legacy driver).

jupiterbjy, 2023-04-27:
Does it show black screen(With backlight) after boot?