goodtft/LCD-show #337: Change xorg Framebuffer to fb1 to support recent Raspbian releases

jupiterbjy, 2023-04-27:
For my [3.5" GPIO Display(MPI3501)](http://www.lcdwiki.com/3.5inch_RPi_Display), changing framebuffer in *xorg* config from `fb0` to `fb1` fixed the issue.

If this wrong framebuffer actually is the only reason why (seemingly) all goodtft displays are displaying blank screen - I suggest to add a patch for it in all driver installation scripts.

---

Following are steps I took to fix the blank (black) screen.
Tested on *Raspberry Pi 1B+ / Raspberry Pi OS (32bit) - release 2023-02-21*

1. Make sure driver is installed correctly
    Check if `dtoverlay` specified in bottom section of `/boot/config.txt/` exists in `/boot/overlays/`.
    ```
    pi@raspberrypi:~/LCD-show/usr $ cat /boot/config.txt | grep ^dtoverlay
    dtoverlay=tft35a:rotate=90
    ```
    
    ```
    pi@raspberrypi:~/LCD-show/usr $ ls -l /boot/overlays/ | grep tft35a
    -rwxr-xr-x 1 root root   2616 Apr 28 03:07 tft35a.dtbo
    -rwxr-xr-x 1 root root   2616 Apr 28 03:45 tft35a-overlay.dtb
    ```

2. Open xorg config
    ```
    sudo vi /usr/share/X11/xorg.conf.d/99-fbturbo.conf
    ```

3. Change framebuffer from 0 to 1 & Save
    ```
    Option "fbdev" "/dev/fb0" 
    ```
    ↓↓↓
    ```
    Option "fbdev" "/dev/fb1"
    ```

4. try running x server on fb1 and see if it works:
    ```
    sudo FRAMEBUFFER=/dev/fb1 startx
    ```

5. Reboot and enjoy
    If it's stuck at white screen, pull out power and reconnect.
    It'll eventually show desktop after long black screen - stuck on wrong rotation though. Touch xy are reversed.
    
    [image]

Above answers are combination of 5-yr-old *RPi forum*'s [post](https://forums.raspberrypi.com/viewtopic.php?t=186996) and *OctoPrint Community*'s [post](https://discourse.zynthian.org/t/new-build-black-screen/6818/14). Took 2 weeks to find but worth enough.

kristian, 2023-08-16:
@jupiterbjy any idea how to make this work with enabling the Full KMS OpenGL drivers in raspi-config? Seems if I enable Full KMS the 99-fbturbo.conf driver is no longer loaded / needed and I don't know how to switch to FB1... I need the full OpenGL drivers to load Midori, which will not work on the legacy drivers for some reason. Any idea? Appreciate any hint.

jupiterbjy, 2023-08-19:
@kristian Ah sadly I do not know the other workaround either - I might have found two [possibilities](https://superuser.com/questions/122173/change-linux-consoles-default-monitor) but I have low hope on these 2 methods.. sorry!

3raz0r, 2023-12-27:
Big THX to your research for fixing this bug. 
I was was really frustrated, the display (in my case a 2,8 inch 320 x 240 cheap ILI9341) worked not regulary on a raspi zero and raspi zero w, mostly he didnt.
An when he worked, the framerate was crap and a black screen showed off after a few minutes (it was not the screensaver). fixing the /usr/share/X11/xorg.conf.d/99-fbturbo.conf file did it for me. This should be fixed for the future. THX again

jupiterbjy, 2023-12-28:
@3raz0r such simple fix never listed on document.. this is quite sad to see, so this is universal problem existing not just my MPI3501 but also ILI9341, thanks for confirmation - I guess now I'm pretty sure all waveshare/goodtft products are experiencing same. Glad it worked for you!

Chocodu31, 2024-12-06:
Hey i follow the step from your edit, and it don't work. I mean, there a black screen but glitchy (when i use the pen i see in a very low framerate the mouse and it's very very glitchy).

Do you think i need to get another screen or it's just that i'm bad at installing driver ?
I follow the edit tutorial and your's, in your's there a error when i do "sudo FRAMEBUFFER=/dev/fb1 startx". And the other get me the weirdly non refresh screen with touch

if you can help me ;c