goodtft/LCD-show #347: MHS-3.5in Raspberry Pi Display Not Working

2lag, 2023-10-16:
Started connected to external display, booted up terminal, cloned repo, changed permissions, and executed the MHS35-show command, waited, and nothing. Almost exactly the same as #336, but for me, I see a cursor in the bottom right of the screen. It does this whether or not I have the external display connected, and won't print text unless I jiggle the Micro-USB in which it will say `[uptime elapsed] hwmon hwmon1: Undervoltage detected!`

goodtft, 2023-10-16:
Is That This Display？

---- Replied Message ----
| From | ***@***.***> |
| Date | 10/16/2023 14:52 |
| To | ***@***.***> |
| Cc | ***@***.***> |
| Subject | [goodtft/LCD-show] MHS-3.5in Raspberry Pi Display Not Working (Issue #347) |

Started connected to external display, booted up terminal, cloned repo, changed permissions, and executed the MHS35-show command, waited, and nothing. Almost exactly the same as #336, but for me, I see a cursor in the bottom right of the screen. It does this whether or not I have the external display connected, and won't print text unless I jiggle the Micro-USB in which it will say [uptime elapsed] hwmon hwmon1: Undervoltage detected!

2lag, 2023-10-16:
Yes of course! It is the MHS 3.5" TFT display for a Raspberry Pi, I am using it with a 3a+ board on Raspbian's latest version, 64-bit. I also tried the latest version with 32-bit and it was the same.

goodtft, 2023-10-16:
Do you have 3B+/4B  or  other RaspberryPi for test?

http://www.lcdwiki.com/MHS-3.5inch_RPi_Display

| |
goodtft
|
|
***@***.***
|
---- Replied Message ----
| From | ***@***.***> |
| Date | 10/16/2023 17:36 |
| To | ***@***.***> |
| Cc | ***@***.***> ,
***@***.***> |
| Subject | Re: [goodtft/LCD-show] MHS-3.5in Raspberry Pi Display Not Working (Issue #347) |

Yes of course! It is the MHS 3.5" TFT display for a Raspberry Pi, I am using it with a 3a+ board on Raspbian's latest version, 64-bit. I also tried the latest version with 32-bit and it was the same.

2lag, 2023-10-16:
I'm using a 3A+!

mukral, 2023-10-18:
I got a similar/same problem with this Display: [http://www.lcdwiki.com/3.5inch_RPi_Display](url)

In may case it's because there is no driver for wayland. It only works with X11
So I had to go to raspi-config and change back to X11 under advanced Settings -> wayland

thinktankbs, 2023-11-13:
I'm having this problem on bullseye with pi4. Change config files and nothing blinking cursor and corrupts the sd card reformatted multiple times just gonna return it get better screen