goodtft/LCD-show #349: LCD does not work

xlxonxe, 2023-10-23:
I am working with a Raspberry Pi 4 and a 3.5 inch RPi Display, with the latest OS. After cloning the display I typed in all the commands listed on the website but now neither my 3.5 nor my HDMI Display are working, both are just black, but the boot. What is going on?

charlygg, 2023-10-23:
Check if you happen to have Raspberry OS 12 bookworm (which is the latest OS right now) installed in your Raspberry PI it would be the reason why the screen is not working.

Your issue is similar to the following https://github.com/goodtft/LCD-show/issues/348

I also tried myself with no success because the new OS 12 bookworm is using Wayland as the new protocol for the Desktop Server, the previous version was using X11 and this repo is based on X11 to work, a way to work around this problem is using the previous version of Raspberry OS 11 bullseye and follow the same steps

oilcan-productions, 2023-11-06:
So using `raspi-config` to switch to X server and re-installing the driver should work on Bookwork, correct?

rar0n, 2024-09-04:
That was a nice theory (switch to X11 using raspi-config). but at least I couldn't get it to work. I'm not sure why, maybe the setting didn't save? (No save option). Also I couldn't get the normal hdmi to work afterwards either.

Tried it on 2024-07-04-raspios-bookworm-arm64-full.

But big thanks to @charlygg for the solution!
Using 2023-05-03-raspios-bullseye-arm64-full, worked!

On a Raspberry Pi 3B