goodtft/LCD-show #350: Not working on Raspberry Pi OS Bookworm

TehseenHasan, 2023-10-26:
I tried it on Raspberry Pi 3 B+ working on Raspberry Pi OS bullseye and it works. But for the latest version of Pi OS "Bookworm" it is not working. Please update the library to support the new OS. Thanks!

olen2006, 2023-10-26:
I have the same issue

bedamichl, 2023-10-30:
I have the same issue on a Pi 4B using Pi OS Bookworm.

linuxlight, 2023-11-05:
I have the same issue on Raspberry Pi 3B+ running Pi OS Bookworm

linuxlight, 2023-11-05:
I have the same issue on Raspberry Pi 3B+ running Pi OS Bookworm.

oilcan-productions, 2023-11-06:
Same on RPI4 with Bookwork 64 bit. Only get a blinking cursor and rotation does not work

ShadyMCShadesen, 2023-11-07:
Same for me. RPI3B with Bookworm 64bit and i only get a blinking cursor, no way to boot to the desktop

lilkideminem, 2023-11-12:
> I have the same issue on RPI4B. After reboot only blinking cursor with no way to boot on desktop. Fixed after using `raspi-config` to switch to X11

this worked!!! thank you!

SeanI-Git, 2023-11-12:
> I have the same issue on RPI4B. After reboot only blinking cursor with no way to boot on desktop. Fixed after using `raspi-config` to switch to X11

I'm using the 3.5" screen with my Pi Zero 2 W. I was initially trying with the latest version of Bookworm 64-bit. In raspi-config, it appeared my display server was already set to X11. Either way, I selected X11, and nothing changed. So I flashed it with the latest version of Bullseye 64-bit, and it worked flawlessly. Not sure what the issue was. Maybe I'll try the Bookworm version one more time. Thank you for the advice on X11 either way!!

SylvainGa, 2023-11-13:
I installed "Raspberry Pi OS (64-bit)" (bookworm) using Raspberry Pie Imager v1.8.1 on my Raspberry Pi 4 and to my surprise, my 4" MPI4008 display came to life on first boot without any modification necessary. I don't have touch available but the screen is working fine as is.

ParzivalSan, 2023-11-19:
> I have the same issue on RPI4B. After reboot only blinking cursor with no way to boot on desktop. Fixed after using `raspi-config` to switch to X11

Sorry, but how do you switch to X11? I can't find the menu. Thanks!

greenpdx, 2023-11-20:
I switch to X but still doesn't work

Raspberry Pi OS with desktop

    Release date: October 10th 2023
    System: 64-bit
    Kernel version: 6.1
    Debian version: 12 (bookworm)
    Size: 1,093MB

andimulk, 2023-12-02:
thankns it's wored

> I have the same issue on RPI4B. After reboot only blinking cursor with no way to boot on desktop. Fixed after using `raspi-config` to switch to X11

it works, but the screen is very big and when I change the screen resolution in sudo raspi-config, the screen remains the same even though the resolution size changes, any suggestions?

Orion-Nebula1500, 2023-12-27:
I have the same issue on my Raspberry pi 4 B

positron96, 2024-01-25:
I saw another problem during installation.
The system date in LCD5-show does not work correctly. The script reads `uname -v` and extracts year from it. Then, if it's less than 2017, some changes are made (not sure what exactly).

Here how `uname` looks in bookworm:
```
root@sx:~# uname -v
#1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24)
root@sx:~# cat /etc/issue
Raspbian GNU/Linux 12 \n \l
```
And here is how it looked before:
```
root@s4:~# uname -v
#1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023
root@s4:~# cat /etc/issue
Raspbian GNU/Linux 11 \n \l
```

The bookworm's year is detected incorrectly it seems.

axwell, 2024-02-14:
same issue here, any updates on the lib ?

pesa1234, 2024-02-15:
Hello, I have raspberry 3B+ with 5" resistive.
Hdmi signal ok with large icons, but I have no response from touch

thuongnv94, 2024-03-13:
have any update? I have switched  to X11 but it still not work

pesa1234, 2024-03-29:
Any news guys? Thanks

ReneXB12X, 2024-04-17:
Waveshare 5 inch HDMI LCD works fine under Bullseye but dies during boot of Bookworm on Raspberry pi 3B+. Answer from Waveshare is that the version 2 of the 5 inch HDMI LCD does not work with Bookworm. The say that you need their version 3.1 but they report still touch problems even with this version.

rene51, 2024-04-19:
Thank you for the suggestion.
I tested 2023-10-10-raspios-bookworm-armhf.img with sudo ./LCD5-show. Prepared the SD card on a headless Pi 3B+ and then booted the SD card on a Pi 3B+ with the Waveshare 5" HDMI LCD. 
The Pi booted with the Raspberry welcome screen but after that it remained black with a fast blinking cursor. SSH into the PI worked and I installed Conky. That opened an X-window on the laptop but the Conky window remained black also.

maxpaine69, 2024-05-08:
any neew? about this?

OldWoman37, 2024-05-19:
See my last couple of posts in issue #369.
https://github.com/goodtft/LCD-show/issues/369