goodtft/LCD-show #368: After this install i can not connect via vnc

abrvalkcplusplus, 2024-04-18:
After this install i can not connect via vnc

abrvalkcplusplus, 2024-04-18:
just close connection. how to recovery old xorg setup?

abrvalkcplusplus, 2024-04-18:
how to remove this shit?

abrvalkcplusplus, 2024-04-18:
sudo ./system_restore.sh 
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Selecting previously unselected package xserver-xorg-input-evdev:armhf.
(Reading database ... 158281 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb ...
Unpacking xserver-xorg-input-evdev:armhf (1:2.10.6-1+b1) ...
dpkg: dependency problems prevent configuration of xserver-xorg-input-evdev:armhf:
 xserver-xorg-input-evdev:armhf depends on libevdev2 (>= 1.2.2+dfsg-1~).
 xserver-xorg-input-evdev:armhf depends on libmtdev1 (>= 1.1.0).
 xserver-xorg-input-evdev:armhf depends on libudev1 (>= 183).
 xserver-xorg-input-evdev:armhf depends on xorg-input-abi-24.
 xserver-xorg-input-evdev:armhf depends on xserver-xorg-core (>= 2:1.18.99.901).

dpkg: error processing package xserver-xorg-input-evdev:armhf (--install):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.11.2-2) ...
Errors were encountered while processing:
 xserver-xorg-input-evdev:armhf
The system has been restored
now reboot

Broadcast message from root@raspberrypi on pts/2 (Thu 2024-04-18 23:48:04 +04):

The system will reboot now!

Broadcast message from root@raspberrypi on pts/2 (Thu 2024-04-18 23:48:04 +04):

The system will reboot now!

abrvalkcplusplus, 2024-04-18:
Linux raspberrypi 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch6

abrvalkcplusplus, 2024-04-18:
this scripts broken vnc connect

abrvalkcplusplus, 2024-04-18:
sudo ./system_restore.sh 
after this vnc is not restored and this small display is not working. no any view desktop
/LCD-show $ sudo ./LCD35-show
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
(Reading database ... 158290 files and directories currently installed.)
Removing xserver-xorg-input-evdev (1:2.10.6-2) ...
Processing triggers for man-db (2.11.2-2) ...
Hit:1 http://deb.debian.org/debian bookworm InRelease                              
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done                              
bad network, can't install cmake!!!
(2023-10-05)
./LCD35-show: line 102: test: (2023-10-05): integer expression expected
need to update touch configuration
Selecting previously unselected package xserver-xorg-input-evdev.
(Reading database ... 158281 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-input-evdev_1%3a2.10.6-2_arm64.deb ...
Unpacking xserver-xorg-input-evdev (1:2.10.6-2) ...
Setting up xserver-xorg-input-evdev (1:2.10.6-2) ...
Processing triggers for man-db (2.11.2-2) ...

reboot now

Broadcast message from root@raspberrypi on pts/2 (Fri 2024-04-19 00:03:45 +04):

The system will reboot now!

just bullshit broken all X display

anuk909, 2024-09-15:
Did you find a way to solve it?