goodtft/LCD-show #369: Require TFT 3.5 display configuration for Raspberry Pi Bookworm release

ruthvikb14, 2024-04-30:
Hi,

I'm raising this issue to request help in installing the TFT3.5 display to Pi 3B+ with Bookworm OS. The current LCD35-show.sh will not work since it is configured to Bullseye and the the new version of Bookworm is > 12.1. And also there are few changes in Bookworm OS like config.txt is shifted from "/boot" to "/boot/firmware" etc. So it is much appreciated if you can include the changes to LCD-show repo to support Bookworm OS

maxpaine69, 2024-05-06:
having the same problem. If i use bullseyes no problems everything perfect, upgrading or new install to Bookworm doesnt show GUI mantains a black screen, and only in console mode it shows the console.

OldWoman37, 2024-05-07:
You need to switch to X11 from wayland using raspi-config.
But you will likely find problems with the touchscreen after.

maxpaine69, 2024-05-07:
> You need to switch to X11 from wayland using raspi-config. But you will likely find problems with the touchscreen after.

already tried didnt work

OldWoman37, 2024-05-07:
Do you have the:
dtoverlay=tft35a:rotate=90
line in your /boot/firmware/config.txt file?
Does your /boot/firmware/overlays folder contain tft35a.dtbo and tft35a-overlay.dtb?
After boot, do you have a "fbcp" process running?

maxpaine69, 2024-05-07:
> Do you have the: dtoverlay=tft35a:rotate=90 line in your /boot/firmware/config.txt file? Does your /boot/firmware/overlays folder contain tft35a.dtbo and tft35a-overlay.dtb? After boot, do you have a "fbcp" process running?

yes to every thing, i even created a new instaltion

OldWoman37, 2024-05-07:
Oh, this may just be a startup thing. Did you have it run the lightdm login screen before or some autologin thing?

maxpaine69, 2024-05-07:
> Oh, this may just be a startup thing. Did you have it run the lightdm login screen before or some autologin thing?

only the autologin on the config, like it as on bullseye,

OldWoman37, 2024-05-07:
Do you have a process "lightdm" or "Xorg" running?

OldWoman37, 2024-05-07:
When running "sudo journalctl -xe", are there services with errors?

maxpaine69, 2024-05-07:
> When running "sudo journalctl -xe", are there services with errors?

not to all the question, no error it strange.

OldWoman37, 2024-05-07:
You may just have to re-run raspi-config and select your settings: 1, S5, B3 or B4.

maxpaine69, 2024-05-07:
> You may just have to re-run raspi-config and select your settings: 1, S5, B3 or B4.

nothing still the same

OldWoman37, 2024-05-07:
You may need to do:
sudo systemctl set-default graphical.target
sudo systemctl enable lightdm
sudo systemctl start lightdm

maxpaine69, 2024-05-07:
> You may need to do: sudo systemctl set-default graphical.target sudo systemctl enable lightdm sudo systemctl start lightdm

nothing still balck screen and blicking

OldWoman37, 2024-05-07:
Can you post the output of: "dmesg | grep fb"

maxpaine69, 2024-05-07:
> Can you post the output of: "dmesg | grep fb"

[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=480 bcm2708_fb.fbheight=320 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:22:4E:3A vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyS0,115200 console=tty1 root=PARTUUID=9cd26e03-02 rootfstype=ext4 fsck.repair=yes rootwait cfg80211.ieee80211_regdom=PT
[    0.060388] raspberrypi-firmware soc:firmware: Firmware hash is 86ccc427f35fdc604edc511881cdf579df945fb4
[    1.711704] bcm2708_fb soc:fb: FB found 1 display(s)
[    1.719710] bcm2708_fb soc:fb: Registered framebuffer for display 0, size 480x320
[   13.915283] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   13.951526] fb_ili9486: module is from the staging directory, the quality is unknown, you have been warned.
[   13.951951] SPI driver fb_ili9486 has no spi_device_id for ilitek,ili9486
[   13.952115] fb_ili9486 spi0.0: fbtft_property_value: regwidth = 16
[   13.952129] fb_ili9486 spi0.0: fbtft_property_value: buswidth = 8
[   13.952140] fb_ili9486 spi0.0: fbtft_property_value: debug = 0
[   13.952149] fb_ili9486 spi0.0: fbtft_property_value: rotate = 90
[   13.952160] fb_ili9486 spi0.0: fbtft_property_value: fps = 30
[   13.952169] fb_ili9486 spi0.0: fbtft_property_value: txbuflen = 32768
[   14.531715] graphics fb1: fb_ili9486 frame buffer, 480x320, 300 KiB video memory, 32 KiB buffer memory, fps=31, spi0.0 at 16 MHz

here it is

OldWoman37, 2024-05-07:
So, to be clear, you have the console screen working, so you can login via a keyboard in textmode, but now, after that systemctl stuff, the screen is black but flashing/blinking?

maxpaine69, 2024-05-07:
the black screen with the flash is a simtop from the begining, i have the boot to desktop from raspi-config.

OldWoman37, 2024-05-07:
I'm sorry, I don't understand the issue then. Can you post a photo of what you see when booting (if not black but some text mode stuff), and then what you get after?

maxpaine69, 2024-05-07:
> I'm sorry, I don't understand the issue then. Can you post a photo of what you see when booting (if not black but some text mode stuff), and then what you get after?

if i configure it to boot to console i see text, no problem there i see the console, but if i configure it to show desktop, it only show a black screen  with a blincking cursor

OldWoman37, 2024-05-07:
Okay. If you connect a USB keyboard, and hold down CTRL-ALT, and then press F1, wait a second, press F2, wait a second, press F3, and so on, will cycle through all the virtual terminals. One of them should show you the graphical login.

maxpaine69, 2024-05-07:
> Okay. If you connect a USB keyboard, and hold down CTRL-ALT, and then press F1, wait a second, press F2, wait a second, press F3, and so on, will cycle through all the virtual terminals. One of them should show you the graphical login.

no i only see a normal console to login not the desktop login.

OldWoman37, 2024-05-07:
Are the other "screens" (not F1) blank with a flashing cursor?

maxpaine69, 2024-05-07:
> Are the other "screens" (not F1) blank with a flashing cursor?

yes only blinking cursor and black screen, even now a cerated a new card with a fresh os.

maxpaine69, 2024-05-08:
it works perfectlly in bullseye, but in bookworm no luck can only use consoel tehre no desktop.

OldWoman37, 2024-05-09:
Let me know the settings you used in the Raspberry Pi Imager, and then the raspi-config options, and I will attempt to replicate the problem here. I think it is just a configuration issue. Later however, you will likely need a new DTB file to make the touch screen work; this is what I had to do.

maxpaine69, 2024-05-09:
> Let me know the settings you used in the Raspberry Pi Imager, and then the raspi-config options, and I will attempt to replicate the problem here. I think it is just a configuration issue. Later however, you will likely need a new DTB file to make the touch screen work; this is what I had to do.

hello ther and thanks, in the iamge i already tried all option of bookworm, the configuration are the normal one ssh username and password, the raspi config option e choose x11, and auto desktop login, the most basic options.

maxpaine69, 2024-05-09:
mean while was experimenting and foun out if i alter this dtoverlay=piscreen,drm i star getting image, but strange colors and touch doesnt work.

OldWoman37, 2024-05-11:
Well, I did everything from the beginning. Ran the Pi Imager with Debian Bookworm w/Desktop (recommended).
After the first boot, I did:

sudo apt update
sudo apt upgrade
sudo reboot

sudo raspi-config
  6, A6, W1
  1, S5, B3
  (it automatically reboots)

git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show
(it reboots)

And it worked. No issues. It seems the patch from 2 months ago to the tft35a.dtb fixes the touchscreen issue. I thought I grabbed that file, but I didn't; had the old one and fixed it for myself.

maxpaine69, 2024-05-11:
right now i only have a  pi0 2w to test i will say something to you when e can getta pi4, i tried on the zero 2w and didnt work, i wil ltry on the pi4

ronaldlw, 2024-05-12:
> Well, I did everything from the beginning. Ran the Pi Imager with Debian Bookworm w/Desktop (recommended). After the first boot, I did:
>
> sudo apt update sudo apt upgrade sudo reboot
>
> sudo [...]

I've been trying to get this to work with both a pi zero w and pi 3b with no luck.

right now i'm working with a 3b and followed the steps you gave exactly and just end up with a black screen with blinking cursor in the upper right of the display.

adjei7, 2024-05-14:
I have the same issue with a Pi 5, but think this is already a known issue. I can get it to work perfectly with a pi 3b+ on Bullseye.

adjei7, 2024-05-14:
OK, this is really weird! I finally got it working and think I know why. 

If I disconnect the main HDMI screen and reboot, then the 3.5inch screen works. If both are connected, then the 3.5 inch acts as a touch pad for the HDMI screen.

OldWoman37, 2024-05-15:
The installation instructions at:
http://www.lcdwiki.com/3.5inch_RPi_Display
kind of imply that you shouldn't be connected to a monitor (ie. do the install over SSH). I just assumed that was the case.
As I understand it, the way these displays work is by using the internal display controller (GPU), but have it output to a smaller resolution, eg. 480x320. Then, the "fbcp" program constantly copies the internal display to the add-on board's frame buffer over the SPI interconnect. This way you can use the internal GPU to produce 3D scenes. If you connect an HDMI cable, you are basically telling the internal controller to switch back to an HD type resolution and then that screws up the X server at some point and perhaps that is why the add-on display won't switch after.

adjei7, 2024-05-15:
You are right. 

Unfortunately, the LCD wiki was only one of the many instruction website that I went to. There was another one that stated (wrongly I guess), that I could either ssh or use a monitor when putting in the instructions. So assumed using monitor was fine. Also, many of the videos, including [this one](https://youtu.be/vCAGzLGTUk4?si=Rir5M-9xCibZfnI5&t=231) suggested that I would see both images at the same time on both monitors.

But hey, it's all good. Hopefully it's working for @maxpaine69

maxpaine69, 2024-05-15:
i already tested no problems to show on pi 4, but im starting to see that pi ist getting away from older version, tryed on a 3 and pi zero 2 w only works until bulleyes, bookworm no luck on both pi zero 2w and pi3.

adjei7, 2024-05-15:
> i already tested no problems to show on pi 4, but im starting to see that pi ist getting away from older version, tryed on a 3 and pi zero 2 w only works until bulleyes, bookworm no luck on both pi zero 2w and pi3.

I will test it out on my pi 3 and let you know.

OldWoman37, 2024-05-15:
> I'm not connecting a monitor and it still doesn't work for me. @OldWoman37 are there any steps you've left out of your process that you said got it working for Debian Bookworm w/Desktop ? What pi is this, i'm using a 3b and have also tried with zero w

I did it on a Pi 4, and I recorded everything I did. But since the distro is the same for 3s and 0s, I don't see why it shouldn't be any different for those.

ronaldlw, 2024-05-15:
> I did it on a Pi 4, and I recorded everything I did. But since the distro is the same for 3s and 0s, I don't see why it shouldn't be any different for those.

Well, I can verify it does work for pi 4 but not 3 or zero

adjei7, 2024-05-16:
> > i already tested no problems to show on pi 4, but im starting to see that pi ist getting away from older version, tryed on a 3 and pi zero 2 w only works until bulleyes, bookworm no luck on both pi zero 2w and pi3.
> 
> I will test it out on my pi 3 and let you know.

I can confirm that this process does not work for Pi 3. Just tested it out using the exact same process as the Pi 5. 

On the Pi 3, after completing the whole process the screen would go black and show only a blinking cursor. Connecting a keyboard would cause an error message (undervoltage detected) to show. Pressing Alt and scrolling though the function keys would show the different terminals F1 showing the main one, but it would jump right back to the blinking cursor after releasing the keys. So the screen works. But the OS doesn't load properly. 

On the Pi 5, after completing the whole process the TFT screen would go black and act as a touch pad for my HDMI screen. When the HDMI screen was disconnected and the pi rebooted, the TFT screen would work perfectly.

I would recommend sticking with Bullseye for Pi 3 and below, as from research there are still a lot of issues with Bookworm.

OldWoman37, 2024-05-16:
Okay. I checked this out on RPi3B+ and I found that a service (glamor-test) was adding a display driver for X. Since X is supposed to use the default driver, this tells it to use a different one, hence the problems you are having. Do the following on a RPi3B+:

sudo systemctl disable glamor-test.service
sudo rm /usr/share/X11/xorg.conf.d/20-noglamor.conf
sudo systemctl restart lightdm

Let me know if that works.

ronaldlw, 2024-05-16:
> Okay. I checked this out on RPi3B+ and I found that a service (glamor-test) was adding a display driver for X. Since X is supposed to use the default driver, this tells it to use a different one, [...]

that does seem to get the display working, thanks!

ronaldlw, 2024-05-17:
> Okay. I checked this out on RPi3B+ and I found that a service (glamor-test) was adding a display driver for X. Since X is supposed to use the default driver, this tells it to use a different one, [...]

Is the script going to be updated to incorporate these fixes?

ronaldlw, 2024-05-17:
here are the updated commands including what is done in raspi-config to disable wayland and enable x:

```
sudo rm /usr/share/X11/xorg.conf.d/20-noglamor.conf
sudo sed -e '/dtoverlay=vc4/ s/^#*/#/' -i /boot/firmware/config.txt
sudo sed -i -e '/greeter-session=/ s/=.*/=pi-greeter/' /etc/lightdm/lightdm.conf
sudo sed -i -e '/user-session=/ s/=.*/=LXDE-pi-x/' /etc/lightdm/lightdm.conf
sudo sed -i -e '/autologin-session=/ s/=.*/=LXDE-pi-x/' /etc/lightdm/lightdm.conf
sudo systemctl disable glamor-test.service
```

Then the rest is the same as normal, but also some of my specific changes if anyone else would be interested:

```
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show

#force the resolution to be 720 x 480 the display scales it and makes it more usable and is still pretty clear
sudo sed -i -e '/hdmi_cvt / s/480 320/ 720 480/' ./LCD35-show

#rotate the display 180 so the power plug is on top instead of the bottom
sudo ./LCD35-show 180
```

OldWoman37, 2024-05-17:
Okay. I have a bit more of an explanation of what is going on here.
If you look at:
https://indico.freedesktop.org/event/2/contributions/60/attachments/66/107/XDC-20220-DRI3-modesetting-without-glamor.pdf
You will see there is a potential issue of running out of memory using a RPi device with lower memory (<1GB). The OpenGL acceleration can use more RAM thus reducing the amount of memory for your CPUs resulting in termination of some programs. Granted you have to be doing quite a bit of stuff for this to happen, but it can happen. So the service "glamor-test" checks for this and disables that acceleration.
The problem for us is that X has a weird way of doing configurations. Before, they had one one file where you put in all your systems info and it just worked. But with todays systems, the configuration is more dynamic and rather than using a single file, we have multiple files, and the order in which they are read can cause problems. This is a known issue, however the X maintainers try to make it work with the most common configurations. Our configuration is not the case, so this "glamor-test" services will work on a stock configuration, but not ours.
So the patch I provided above is good as long as you keep what you are running to a minimum (ie. a few tabs in a browser). If you aren't running a browser at all, you will probably be fine. Otherwise, you can do this:

```
sudo systemctl disable glamor-test.service
sudo cat /usr/share/X11/xorg.conf.d/20-noglamor.conf | sed -e '/EndSection/i Driver "fbdev"' > /usr/share/X11/xorg.conf.d/21-noglamor.conf
sudo rm /usr/share/X11/xorg.conf.d/20-noglamor.conf
sudo systemctl restart lightdm
```

What is different he from the previous solution is that OpenGL is still disabled, but I the display driver is specified after (ie. "fbdev") which results in X starting up properly. So this is a better choice if you will run heavier applications and don't want to risk loosing too much RAM to the OpenGL system.
One warning though: if you move your installation (SD card) into another system, like a RPi4 or better, you will need to re-enable the glamor-test service as well as delete the 21-noglamor.conf file.

maxpaine69, 2024-05-18:
> Do you have the: dtoverlay=tft35a:rotate=90 line in your /boot/firmware/config.txt file? Does your /boot/firmware/overlays folder contain tft35a.dtbo and tft35a-overlay.dtb? After boot, do you have a "fbcp" process running?

yes to every thing, i even created a new instaltion

> sudo systemctl restart lightdm

resulted on the pi zero 2w.

OldWoman37, 2024-05-18:
Did you do what I posted yesterday?

> yes to every thing, i even created a new instaltion
>
>
> resulted on the pi zero 2w.

Did you do what I posted yesterday?

mayurnewase, 2024-05-25:
had same issue with 3B+ and Bookworm, 
ran commands from both these comments

https://github.com/goodtft/LCD-show/issues/369#issuecomment-2115596683
https://github.com/goodtft/LCD-show/issues/369#issuecomment-2116552991

then reinstalled the drivers again, then it screen started working.
Thanks @OldWoman37 @ronaldlw

Elmiar0642, 2024-06-02:
> Okay. I have a bit more of an explanation of what is going on here.
> If you look at:
> https://indico.freedesktop.org/event/2/contributions/60/attachments/66/107/XDC-20220-DRI3-modesetting-without-glamor.pdf
> You will see there is a potential issue of running out of memory using a RPi device with lower memory (<1GB). The OpenGL acceleration can use more RAM thus reducing the amount of memory for your CPUs resulting in termination of some programs. Granted you have to be doing quite a bit of stuff for this to happen, but it can happen. So the service "glamor-test" checks for this and disables that acceleration.
> The problem for us is that X has a weird way of doing configurations. Before, they had just one file where you put in all your systems info and it just worked. But with todays systems, the configuration is more dynamic and rather than using a single file, we have multiple files, and the order in which they are read can cause problems. This is a known issue, however the X maintainers try to make it work with the most common systems. Our configuration is not the case, so this "glamor-test" services will work on a stock configuration, but not ours.
> So the patch I provided above is good as long as you keep what you are running to a minimum (ie. a few tabs in a browser). If you aren't running a browser at all, you will probably be fine. Otherwise, you can do this:
> 
> ```
> sudo systemctl disable glamor-test.service
> sudo cat /usr/share/X11/xorg.conf.d/20-noglamor.conf | sed -e '/EndSection/i Driver "fbdev"' > /usr/share/X11/xorg.conf.d/21-noglamor.conf
> sudo rm /usr/share/X11/xorg.conf.d/20-noglamor.conf
> sudo systemctl restart lightdm
> ```
> 
> What is different he from the previous solution is that OpenGL is still disabled, but I the display driver is specified after (ie. "fbdev") which results in X starting up properly. So this is a better choice if you will run heavier applications and don't want to risk loosing too much RAM to the OpenGL system.
> One warning though: if you move your installation (SD card) into another system, like a RPi4 or better, you will need to re-enable the glamor-test service as well as delete the 21-noglamor.conf file.

After doing this the screen boots up, but even after setting boot up to GUI, the screen starts in tty1 in terminal. I created even xinitrc and checked. Gnome or cinnamon doesn't open. Why is that?

ld3300, 2024-07-05:
> had same issue with 3B+ and Bookworm, ran commands from both these comments
> 
> [#369 (comment)](https://github.com/goodtft/LCD-show/issues/369#issuecomment-2115596683) [#369 (comment)](https://github.com/goodtft/LCD-show/issues/369#issuecomment-2116552991)
> 
> then checked processes running, `fbcp` wasn't in the list, so reinstalled the drivers again, then it screen started working. Thanks @OldWoman37 @ronaldlw

Thank you for referencing these links, made all the difference for me.
Also thanks @OldWoman37 and @ronaldlw

Jerry12798, 2024-08-13:
> Okay. I have a bit more of an explanation of what is going on here.
> If you look at: [...]

I'm a bit confused of what's happening with the second command here. It's also not running properly for me for a permissions error. I got my MHS35 to work but I'm trying to understand this. So I'm thinking that this is adding "fbdev" to some place in 20-noglamor.conf then copying the file into 21-noglamor.conf?

Jerry12798, 2024-08-13:
> here are the updated commands including what is done in raspi-config to disable wayland and enable x:
>
> ```
> sudo rm /usr/share/X11/xorg.conf.d/20-noglamor.conf
> sudo sed -e '/dtoverlay=vc4/ s/^#*/#/' [...]

Hey I got my MHS35 working with this but I want to understand a bit more so I know why it worked. Would you mind explaining what the sed commands are doing? Also should these commands be added into the MHS35-show script? I tucked them in just above the reboot command

Jerry12798, 2024-08-13:
> > Okay. I have a bit more of an explanation of what is going on here.
> > If you look at:
> > https://indico.freedesktop.org/event/2/contributions/60/attachments/66/107/XDC-20220-DRI3-modesetting-without-glamor.pdf
> > You will see there is a potential issue of running out of memory using a RPi device with lower memory (<1GB). The OpenGL acceleration can use more RAM thus reducing the amount of memory for your CPUs resulting in termination of some programs. Granted you have to be doing quite a bit of stuff for this to happen, but it can happen. So the service "glamor-test" checks for this and disables that acceleration.
> > The problem for us is that X has a weird way of doing configurations. Before, they had just one file where you put in all your systems info and it just worked. But with todays systems, the configuration is more dynamic and rather than using a single file, we have multiple files, and the order in which they are read can cause problems. This is a known issue, however the X maintainers try to make it work with the most common systems. Our configuration is not the case, so this "glamor-test" services will work on a stock configuration, but not ours.
> > So the patch I provided above is good as long as you keep what you are running to a minimum (ie. a few tabs in a browser). If you aren't running a browser at all, you will probably be fine. Otherwise, you can do this:
> > 
> > ```
> > sudo systemctl disable glamor-test.service
> > sudo cat /usr/share/X11/xorg.conf.d/20-noglamor.conf | sed -e '/EndSection/i Driver "fbdev"' > /usr/share/X11/xorg.conf.d/21-noglamor.conf
> > sudo rm /usr/share/X11/xorg.conf.d/20-noglamor.conf
> > sudo systemctl restart lightdm
> > ```
> > 
> > What is different he from the previous solution is that OpenGL is still disabled, but I the display driver is specified after (ie. "fbdev") which results in X starting up properly. So this is a better choice if you will run heavier applications and don't want to risk loosing too much RAM to the OpenGL system.
> > One warning though: if you move your installation (SD card) into another system, like a RPi4 or better, you will need to re-enable the glamor-test service as well as delete the 21-noglamor.conf file.
> 
> I'm a bit confused of what's happening with the second command here. It's also not running properly for me for a permissions error. I got my MHS35 to work but I'm trying to understand this. So I'm thinking that it is adding "fbdev" to some place in 20-noglamor.conf then copying the file into 21-noglamor.conf? My pi doesn't have 20noglamor.conf.. also I'm not sure if these are a run and done or should I add them into the MHS35-show script.

OldWoman37, 2024-08-13:
> I'm a bit confused of what's happening with the second command here. It's also not running properly for me for a permissions error. I got my MHS35 to work but I'm trying to understand this. So I'm [...]

The second line (sed commands) simply puts ' Driver "fbdev" ' in before the ' EndSection ' line. This is to ensure that the configuration file is correct for the X server.
I remove the 20-noglamor.conf file so that updates don't remove the changes made by the sed command; it is now 21-noglamor.conf.
I know this isn't the best solution, but it will work fine for most running the Bookworm.

Jerry12798, 2024-08-13:
> The second line (sed commands) simply puts ' Driver "fbdev" ' in before the ' EndSection ' line. This is to ensure that the configuration file is correct for the X server.
>
> I remove the [...]

Ok I see. I believe that I ran the commands from your first post before I went to do these here in this second version you posted that I replied to. Therefore my 20-noglamor.conf was previously deleted causing the error when attempting to run this sed command. Is 20-noglamor.conf important? I have no issues not having it and everything seems to be fine. The only issue I'm seeing is that previously and I'm talking 2021 or 2022 when I first got this screen it was needed to run the command to switch from the 3.5 touchscreen to the hdmi jack however now it seems both are working simultaneously which I'm not so sure how I feel about.

OldWoman37, 2024-08-13:
> Ok I see. I believe that I ran the commands from your first post before I went to do these here in this second version you posted that I replied to. Therefore my 20-noglamor.conf was previously [...]
No, the 20-noglamor.conf is not important.
The 3.5" screen is a mirror of the HDMI output. It was changed to this so that the 3.5" can benefit from the 3D/hardware acceleration of the internal display. It is also a lot easier to maintain the code base when newer Linux kernels and X servers are released. Previously it had it's own driver, but as things change, and they change often, that driver would have to be fixed, and fixed, and it would vary for every type of external display. This makes things easier in general. However, from what I hear, it may not last as the Linux/ X server developers are deprecating the API layers that allow all this to work. You may want to be careful before you do major OS upgrades.

Jerry12798, 2024-08-14:
> No, the 20-noglamor.conf is not important.
>
> The 3.5" screen is a mirror of the HDMI output. It was changed to this so that the 3.5" can benefit from the 3D/hardware acceleration of the internal [...]

Thanks again for all the help and information! For people coming here in the future should they run their commands after doing LCD35-show/MHS35-show or rather add the commands into the toggle on script itself? I'm asking because I think we should make a pull requests with this fix. I will note this all down.. but another question I have is has this been updated to have the same behavior on Ubuntu? The reason being is I downloaded a Ubuntu preloaded image from the LCD Wiki and it had the old functionality of only working on one display at a time. In fact I have not been able to switch to my monitor or should I say the LCD-hdmi does not seem to work.

OldWoman37, 2024-08-15:
> Thanks again for all the help and information! For people coming here in the future should they run their commands after doing LCD35-show/MHS35-show or rather add the commands into the toggle on [...]

I'm not in any way affiliated with this project (LCD-show) or Ubuntu or Raspberry Pi. I just had a problem with my Pi after updating to bookwork with the 3.5" display and I did a deep dive on what went wrong and how to fix it. I worked with some people in this bug report about it and came up with a solution. So it is up to whomever to integrate this properly into a better overall solution.
Your older "image" probably runs an earlier version of Linux where there was a non-mirror driver for this board. As stated earlier, those APIs are being deprecated and won't work in new releases. I'm sure our problems with these 3.5" displays are not a concern to the major developers. We make due with what we have.
You can keep using an older image, but at some point, it won't be supported and if your system is on the net, then it may be vulnerable so it may be better to stick with newer than what was "better" before.

JAQuent, 2024-08-31:
> Well, I did everything from the beginning. Ran the Pi Imager with Debian Bookworm w/Desktop (recommended). After the first boot, I did:
>
> sudo apt update sudo apt upgrade sudo reboot
>
> sudo [...]

I use a 3.5 inch LCD screen for a Raspberry PI 4 and followed @OldWoman37's steps and it worked without a problem.

mowdep, 2024-12-02:
> here are the updated commands including what is done in raspi-config to disable wayland and enable x:
>
> ```
> sudo rm /usr/share/X11/xorg.conf.d/20-noglamor.conf
> sudo sed -e '/dtoverlay=vc4/ s/^#*/#/' [...]

I tested it and it just work fine on PiZeroW1 with Bullseye ! Thanks you man !

[3 more replies]