goodtft/LCD-show #372: Low refresh rate on MHS-3.5” RPi Display

hellozzl0, 2024-05-16:
Hi there,
  I've got this 3.5 inch DIsplay which the wiki said it can run about 50Hz in retropi([http://www.lcdwiki.com/MHS-3.5inch_RPi_Display](url)), but I got a poor performance on my pi5. 
  It got terrible tearing when moving the window whether in X or wayland, and the es2gears also could'nt run smoothly. Then I tried just memset the fb and it can only execute about 20 times in a second. 

This is the info reported by demsg:
[    3.280359] fb_ili9486: module is from the staging directory, the quality is unknown, you have been warned.
[    3.280447] SPI driver fb_ili9486 has no spi_device_id for ilitek,ili9486
[    3.280538] fb_ili9486 spi0.0: fbtft_property_value: regwidth = 16
[    3.280541] fb_ili9486 spi0.0: fbtft_property_value: buswidth = 8
[    3.280544] fb_ili9486 spi0.0: fbtft_property_value: debug = 0
[    3.280546] fb_ili9486 spi0.0: fbtft_property_value: rotate = 270
[    3.280548] fb_ili9486 spi0.0: fbtft_property_value: fps = 30
[    3.280550] fb_ili9486 spi0.0: fbtft_property_value: txbuflen = 32768
[    3.776298] graphics fb1: fb_ili9486 frame buffer, 480x320, 300 KiB video memory, 32 KiB buffer memory, fps=30, spi0.0 at 115 MHz

OldWoman37, 2024-05-19:
Just curious if you are using /dev/fb0 or /dev/fb1 for your display device?
The drivers for MHS-35 is not just a driver, but also a screen mirroring tool. By default is mirrors the screen (the HDMI output ie. /dev/fb0) to the LCD screen (ie. /dev/fb1). That probably isn't ideal for your application. You  probably want to setup retropi to use /dev/fb1 instead. Also you should also remove (or rename) /dev/rc.local to stop the mirroring on startup (process "fbcp").

hellozzl0, 2024-05-20:
> Just curious if you are using /dev/fb0 or /dev/fb1 for your display device? The drivers for MHS-35 is not just a driver, but also a screen mirroring tool. By default is mirrors the screen (the HDMI [...]

/dev/fb0 is setup by cmdline.txt used to force hdmi output. Just like fbcp,  I implement a mirroring tool in order to copy frame to fb1. I dead this because DispmanX is not available in pi5. And I'm using sway with wayland backend, the script will not working for it, so I just copy the dtbo file. And after several days of testing, it seems the fb1 write speed is slow. It performs perfectly when I downgrade the resolution to 320x160 (half width&height).

OldWoman37, 2024-05-20:
> /dev/fb0 is setup by cmdline.txt used to force hdmi output. Just like fbcp, I implement a mirroring tool in order to copy frame to fb1. I did this because DispmanX is not available in pi5. And I'm using sway with wayland backend, the script will not working for it, so I just copy the dtbo file. And after several days of testing, it seems the fb1's write speed is slow. It performs perfectly when I downgrade the resolution to 240x160 (half width&height).

Yeah, this is a problem in general for fbcp and pi5s. As for the screen tearing, etc, that is expected with an SPI display; the "S" is for serial, so you can't expect to have 50Hz output at that resolution; there has to be a compromise. You can try to increase the speed of the SPI transfers (see fbcp website), or lower the resolution.
On the plus side, since the Pi5s do have a better expansion bus (pcie), someone is going to probably release a faster way to run these displays.

hellozzl0, 2024-05-21:
> Yeah, this is a problem in general for fbcp and pi5s. As for the screen tearing, etc, that is expected with an SPI display; the "S" is for serial, so you can't expect to have 50Hz output at that [...]

Yes, seems problem is the display it self, I also have a 2.2 inch 320x240 SPI display, it can reach more than 30fps with specifying 48MHz. I also have tried the tinydrm driver, it can reduce screen tearing, but still has a low refresh rate, especially when performing whole screen update.
[image]
[image]
SPI frequency has been set to 125MHz

closed 2024-05-21 by hellozzl0

Styne13, 2024-09-15:
Hi @hellozzl0,

try to get the MHS-35B working with wayland and sway, and saw you got it working basically.

Could you please explain a little further here:
„And I'm using sway with wayland backend, the script will not working for it, so I just copy the dtbo file.“

Which OS are you using?
I got the screen working in PiOS lite, but when I start sway, it complains about the gpu beeing not present.
In PiOS Desktop I only got a black screen with a blinking cursor afer executing MHS35B-show.

Thanks!

Marcel

Styne13, 2025-03-30:
Just because it might be helpful for documentation for other people:
I got it working with piscreen driver, latest bookworm and wayland.

After an update
`sudo apt update && sudo apt upgrade -y`

I added the following to /boot/firmware/config.txt:
`dtparam=spi=on
dtoverlay=piscreen,drm,fps=50`
(You can also add screen rotation here if you like)

Now wayland session works, but the refresh-rate is also not the best.