goodtft/LCD-show #382: Using the wrong config.txt file on Raspberry Pi OS - Bookworm

AndrewFromMelbourne, 2024-08-21:
/boot/config.txt was previously a symbolic link to /boot/firmware/config.txt , but not any more.

Your scripts need to check to see if /boot/firmware/config.txt exists and use that if it does.

GreenCrackerDK, 2024-08-28:
Could you provide me with the code or perhaps a costum script? It removed my audio completely after uninstalling the screen and using lcd-hdmi script.

AndrewFromMelbourne, 2024-08-28:
@GreenCrackerDK I was just reporting a problem. I don't have the solution to your problem. Perhaps try asking on the Raspberry Pi forum.

AafaaqAli, 2024-09-14:
were you able to fix it @AndrewFromMelbourne ? 

Device -> RasberryPi 4 4GB 
OS -> Rasbian Bookworm
Display SKU:  MPI3501

AndrewFromMelbourne, 2024-09-15:
@AafaaqAli, I didn't try. I was just reporting this because I was trying to help someone from the Raspberry Pi forum and found this issue. It is really up to @goodtft to fix.