goodtft/LCD-show #393: Joy-IT RB-TFT3.2-V2 on RaspiOS Bookworm

WieMi72, 2025-02-13:
Hello, I'm trying to get this display to work on Bookworm but I can't.
Do you have a solution?
I only get a blinking cursor as a display.

OS:
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

Raspi:
Raspberry Pi 3 Model B Rev 1.2

Thanks for Info and Help.

datviNasser, 2025-02-14:
I got exakt the same problem. Tried a lot, but nothing worked 
(you can connect a keyboard and press ESC / Space in this condition - it then shows some Inits, but won't boot into the OS)

datviNasser, 2025-02-25:
Had same issue with **TFT 3.5"**, could only solve it with: https://joy-it.net/en/products/RB-TFT3.5, scroll down to downloads and get an older OS version: "Preinstalled Raspbian system from 19.12.2018" -> opens a drive link from Joy-IT 

[https://joyiteurope-my.sharepoint.com/personal/onedrive_joyiteurope_onmicrosoft_com/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fonedrive%5Fjoyiteurope%5Fonmicrosoft%5Fcom%2FDocuments%2F2018%2D12%2D19%2Draspbian%2Djessie%5FTFT35%5F800x600%5FHDMI%2Ezip&parent=%2Fpersonal%2Fonedrive%5Fjoyiteurope%5Fonmicrosoft%5Fcom%2FDocuments&ga=1]

Big111156, 2025-02-28:
it seems that these problems are based on the change to wayland and labwc  instead of X11 and openbox...

hope there is soon a solution for the install scripts....

KlausDerOptiker, 2025-03-15:
Hi friends!
do have somehow the same problem (Raspi zero 2 w, 5 inch HDMI LCD (B) 800x480 Pix, 'git clone https://github.com/goodtft/LCD-show.git', running LCD5-show). I investigated a little bit and found out that the shift of cmdline.txt and config.txt to "/boot/firmware" in "bookworm" might be a master problem. I suggest, that a developer able to adjust takes over from here... --- yes and please: I adjusted my bookworm to use X11. Please never switch back to wayland in your setup (I could see while booting that there has been a try to install wayland...! This run into a timeout. (I need x11 because of RealVNC not yet capable of Wayland.)
Hope to see a quick reaction. My knowledege is a bit limited in programming bash.

KlausDerOptiker, 2025-03-15:
Okay -- I saw that I should use MPI5001... same result.

Tunikas, 2025-05-04:
I tried out a lot but i have the same issue with the 3.5 display. 

Some instruction recommended to use a bigger software version 

[image]

but no use. I tried the image recommened by the seller. no use

I looked into the Xorg.0.log and Syslogs played around with the lightdm conf etc.

The best results i actually had with the newest version bookworm. That at least got me a logo upon boot and a blinking cursor, the other OS versions mostly didn't show a logo and didn't even show a cursor.

f-o-n, 2025-05-10:
Also had the same issues. Here a quick writeup how i got it working: #396