goodtft/LCD-show #396: Getting the 3.5" Display to work with moode audio and the Rasberry Pi OS 12 Bookworm

f-o-n, 2025-05-10:
The current script did not work with the latest Moode Raspberry Pi Bookworm release and my Raspberry Pi 4B with an MPI3501 display. Here are the manual steps I took to get it working together with Moode Audio:

### 1. **Install missing (and legacy) packages:**

```bash
sudo apt install xserver-xorg-input-evdev xserver-xorg-video-fbdev xserver-xorg-video-fbturbo lightdm evtest git
```

### 2. **Switch from Wayland to X11:**

Run:

```bash
sudo raspi-config
```

Navigate to:
**6 Advanced Options → A6 Wayland → W1 X11**

### 3. **Get files from repo:**

```bash
git clone https://github.com/goodtft/LCD-show
cd LCD-show
sudo mkdir -p /etc/X11/xorg.conf.d
sudo cp ./usr/tft35a-overlay.dtb /boot/overlays/tft35a.dtbo
sudo cp ./usr/99-calibration.conf-35-90 /etc/X11/xorg.conf.d/99-calibration.conf
sudo cp ./usr/99-fbturbo.conf /usr/share/X11/xorg.conf.d/99-fbturbo.conf
```

### 4. **Edit `/usr/share/X11/xorg.conf.d/99-fbturbo.conf`:**

```ini
Section "Device"
    Identifier      "Allwinner A10/A13/A20 FBDEV"
    Driver          "fbturbo"
    # Switch to /dev/fb0 for default output (usually hdmi) with evdev
    Option          "fbdev" "/dev/fb1"
    Option          "SwapbuffersWait" "true"
    Option          "DRI" "false"
    Option          "ShadowFB" "off"
EndSection
```

### 5. **Edit `/etc/X11/xorg.conf.d/99-calibration.conf`:**

```ini
Section "InputClass"
    Identifier      "calibration"
    MatchProduct    "ADS7846 Touchscreen"

    # Using libinput (better touch support, matrix-based calibration)
    Driver "libinput"
    # Option "TransformationMatrix" "-1 0 1 0 -1 1 0 0 1"

    # Or use evdev (worse touch support in chrome but supports direct calibration)
    # Driver "evdev"
    # Option "Calibration" "3936 227 268 3880"
    # Option "SwapAxes" "0"
    # Option "TransformationMatrix" "1 0 0 0 -1 1 0 0 1"
    # Option "EmulateThirdButton" "1"
    # Option "EmulateThirdButtonTimeout" "750"
    # Option "EmulateThirdButtonMoveThreshold" "30"
EndSection
```

### 6. **Touchscreen calibration (only if using evdev):**

```bash
sudo apt install xinput-calibrator
DISPLAY=:0 xinput_calibrator
```

Results can then be persisted in `99-calibration.conf`

### 7. **Add or modify following Values in `/boot/firmware/config.txt`:**

```ini
dtparam=spi=on
enable_uart=1
dtoverlay=tft35a:rotate=0

hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt 320 480 60 6 0 0 0
hdmi_drive=2

# Important: keep this line present but commented to pass screen size to Chromium with moode
#dtoverlay=vc4-kms-v3d
```

### 8. **Enable autologin with LightDM:**

Edit `/etc/lightdm/lightdm.conf`:

```ini
[Seat:*]
autologin-user=admin  # or your username
autologin-user-timeout=0
```

### Notes:

* Moode's UI scales better in portrait mode; hence the resolution is set to **320x480** and rotation to **0**.
* If you rotate the screen, you must also set the corresponding **transformation matrix** for the touchscreen.
  This tool is helpful: [https://codepen.io/GottZ/full/pWpNgK](https://codepen.io/GottZ/full/pWpNgK)
* When running display config commands over SSH, prepend them with `DISPLAY=:0`.

### Debugging Tools:

```bash
cat /var/log/Xorg.0.log | grep EE
cat /var/log/Xorg.0.log | grep fb
ls /dev/fb*
fbset -fb /dev/fb0
sudo evtest
```

### Moode Settings:

In Moode UI:
Go to **Configure → Peripherals → General settings** and enable **Display**.
This ensures the display manager is started and `.xinitrc` is executed, which launches Chromium in kiosk mode with the Moode UI.

### References:

* [https://michael-peeters.blogspot.com/2018/03/](https://michael-peeters.blogspot.com/2018/03/)
* [https://github.com/jwa-7/goodtft-kiosk](https://github.com/jwa-7/goodtft-kiosk)

jasperkuipers, 2025-05-13:
Hi!

This look promising! Do you think this workflow will work for the MHS35 as well? Are there any mutation needed?

Like to hear from you!

Thanks in advance

Jasper

f-o-n, 2025-05-14:
Probably, haven't tested it but you can try changing following:

Copy the right overlay for your display: `sudo cp ./usr/mis35-overlay.dtb /boot/overlays/mis35.dtbo`
and in `/boot/firmware/config.txt` the overlay and the right resolution needs to be selected:
```
dtoverlay=mis35:rotate=90
hdmi_cvt 480 320 60 6 0 0 0"
```

`/etc/X11/xorg.conf.d/99-calibration.conf` also needs the correct calibration matrix if evdev is used.

jasperkuipers, 2025-05-14:
@f-o-n , thank you for your reply. I will test it out and let you know if it worked. Do you also know how I can set this touchscreen as a single display? Currently it forces me to connect a HDMI monitor as well.