goodtft/LCD-show #404: compilation terminated on new raspberryos trixie

MrTreamer, 2025-08-14:
Package libraspberrypi-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libdtovl0:armhf libdtovl0

E: Package 'libraspberrypi-dev' has no installation candidate
...
[ 50%] Building C object CMakeFiles/fbcp.dir/main.c.o
/home/.../LCD-show/rpi-fbcp/main.c:9:10: fatal error: bcm_host.h: No such file or directory
    9 | #include <bcm_host.h>
      |          ^~~~~~~~~~~~
compilation terminated.

and

the install process deletes packages depending with "raspi-config"

after reinstall "raspi-config" no LCD output avaible

PyroFourTwenty, 2025-09-25:
Maybe i should have been more precise in explaining what i did. I am using a 3.2" display. After a couple times of flashing a fresh install of Raspberry Pi OS i found that installing the packaged mentioned earlier PRIOR to installing the drivers (via `sudo ./LCD32-show`) fixes the compilation process happening in the `LCD32-show` script. 

I propose you try the following:

0. (if possible) try to reinstall your Raspberry Pi OS
1. `sudo apt update && sudo apt upgrade`
2. `sudo apt install apt-utils build-essential -y` (do this before running your respective driver installation script)
3. install the driver (for me its the 3.2" LCD) 
```
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD32-show
```

katzenjens, 2025-10-18:
'libraspberrypi-dev' is no longer available in trixie, so fbcp can't be built. As there is no replacement for fbcp, you have to switch the framebuffer manually with con2fbmap. I did some research and put it here: https://github.com/katzenjens/lcd32 It works with 3.2 and 3.5 inch tft on console and KlipperScreen on every Raspberry Model excluding the Raspberry 5.

de-finnik, 2025-11-05:
Works like a charme, thanks @katzenjens