goodtft/LCD-show #408: PI 5 16G - PI OS (64-bit) - Unable to start Session after installed

mjdusa, 2025-09-13:
After performing the following:

sudo rm -rf LCD-show

git clone https://github.com/goodtft/LCD-show.git

chmod -R 0755 LCD-show

cd LCD-show/

sudo ./MHS35-show

reboot

Disconnecting external monitor while rebooting.

Login screen shows up on LCD fine aft a longer then normal boot.

Attempting to login in to any user returns:
"Failed to start session"

Have tried and get the same with the original external monitor attached.

mjdusa, 2025-09-13:
This is similar, but slightly different from https://github.com/goodtft/LCD-show/issues/399

katzenjens, 2025-10-17:
The issue lies in fbcp. It relies on obsolete libs. And there is no alternative available at the moment. Since fbcp just copies the framebuffer, instead the fb must be toggled via con2fbmap. I ripped apart the howto and started quick-n-dirty from scratch. In the end, it is less complicated. You may try my repo https://github.com/katzenjens/lcd32 I have only included the 3.2 and 3.5 inch display. Both work with PI 1, PI Zero up to PI4. Since I have no PI5 available, i don't know if it works and if it needs more tweaks.